lvgl/lvgl #8359: Minor Bug in Code that Generates CHANGELOG List

vwheeler63, 2025-06-04:
### LVGL version

v9.3

### Platform

The platform that generates the CHANGELOG list before each release.

### What happened?

In the code that generates the list for `CHANGELOG.rst`, something is either adding, or failing to remove, a space that is causing problems.  I saw this right after last LVGL version release as well and thought it was somebody hand-typing, and so didn't report it then.  Now I know it is auto-generated.

A specific example of it is in `CHANGELOG.rst` line 70:
```rst
- **feat(display): add triple buffer support ** `8158 <https://github.com/kisvegabor/lvgl_upstream/pull/8158>`__
```
As you can see, the whole line is boldfaced because the parser sees the leading `**`, but then does not see the trailing `**` because there is a space in front of it.  To be parsed correctly, that space has to be removed.  This is what the parser sees after it is removed.
```rst
- **feat(display): add triple buffer support** `8158 <https://github.com/kisvegabor/lvgl_upstream/pull/8158>`__
```
See the difference?

In any case, whatever generates that list should be tweaked to remove that space.  There were 8 of these on the following lines, and this is the warnings that Sphinx generates when it sees this:
```text
C:\TEMP\lvgl_docs\CHANGELOG.rst:70: WARNING: Inline strong start-string without end-string. [docutils]
C:\TEMP\lvgl_docs\CHANGELOG.rst:97: WARNING: Inline strong start-string without end-string. [docutils]
C:\TEMP\lvgl_docs\CHANGELOG.rst:121: WARNING: Inline strong start-string without end-string. [docutils]
C:\TEMP\lvgl_docs\CHANGELOG.rst:389: WARNING: Inline strong start-string without end-string. [docutils]
C:\TEMP\lvgl_docs\CHANGELOG.rst:467: WARNING: Inline strong start-string without end-string. [docutils]
C:\TEMP\lvgl_docs\CHANGELOG.rst:844: WARNING: Inline strong start-string without end-string. [docutils]
C:\TEMP\lvgl_docs\CHANGELOG.rst:885: WARNING: Inline strong start-string without end-string. [docutils]
C:\TEMP\lvgl_docs\CHANGELOG.rst:967: WARNING: Inline strong start-string without end-string. [docutils]
```

I don't know exactly where this code is, but if someone would point that out, I'd be happy to address it.

### How to reproduce?

Run the code that generates the CHANGELOG list.

kisvegabor, 2025-06-11:
Thank for reporting it! I've fixed the links from v8.4  to v9.3.

kisvegabor, 2025-07-14:
Which "title" do you mean?

vwheeler63, 2025-07-14:
> Which "title" do you mean?

The reason I wrote this is that the original CHANGELOG.rst file I dealt with had contents that looked like this:

```rst
New Features
~~~~~~~~~~~~

- **feat(wayland): add  dmabuf support with g2d** `8122 <https://github.com/lvgl/lvgl/pull/8122>`__
- ** feat(draw_sw): add RGB565_SWAPPED support** `8227 <https://github.com/lvgl/lvgl/pull/8227>`__
- **feat: add new lvgl examples ** `8225 <https://github.com/lvgl/lvgl/pull/8225>`__
- ...etc.
```

with occasional space after the initial `**`, and an occasional space before the 2nd `**` as illustrated on items 2 and 3 in the list above.  That causes Sphinx reST parsing to NOT convert that string to boldface, plus it (correctly) generates a Sphinx warning when it happens.  I'm calling that string between the `**` the "title" -- title for each line.

If it isn't already done, something needs to detect and remove those leading and trailing spaces.

kisvegabor, 2025-07-28:
Oh I see. The related code is 
- https://github.com/lvgl/lvgl/blob/master/scripts/changelog_gen.sh
- https://github.com/lvgl/lvgl/blob/master/scripts/changelog-template.hbs
- 
But I don't know if we can trim the white spaces. 🙁

vwheeler63, 2025-08-12:
Not stale