lvgl/lvgl #8375: LV_OBJ_FLAG_OVERFLOW_VISIBLE with lv_obj_update_layout  display is not normal

KioZhang, 2025-06-06:
### LVGL version

v9.3.0-43-g9cf1e38a9

### Platform

I use lv_port_linux with SDL on linux

[main.c](https://github.com/user-attachments/files/20620702/main.c.txt)
[lv_conf.h](https://github.com/user-attachments/files/20620701/lv_conf.h.txt)

### What happened?

The object hierarchy is as follows
```
parent
    ├── layer
    └── contparent
               └── cont
```

When I called `lv_obj_update_layout` on the `layer` object, it invalidated the `LV_OBJ_FLAG_OVERFLOW_VISIBLE` attribute of `contparent` and `cont`.
I actually found that the problem was caused by one of the objects calling `lv_obj_align_to`, which invalidated the `LV_OBJ_FLAG_OVERFLOW_VISIBLE` property, and then I realized that it was caused by `lv_obj_update_layout`.

normal：

[image]

error：

[image]

### How to reproduce?

``` c
static void __obj_set_ext_draw_size_cb(lv_event_t * e)
{
    int32_t s = (int32_t)(intptr_t)lv_event_get_user_data(e);
    lv_event_set_ext_draw_size(e, s);
}

void lv_obj_set_overflow_visible_flag(lv_obj_t * obj, int32_t size)
{
    lv_obj_add_flag(obj, LV_OBJ_FLAG_OVERFLOW_VISIBLE);
    lv_obj_add_event_cb(obj, __obj_set_ext_draw_size_cb, LV_EVENT_REFR_EXT_DRAW_SIZE, (void *)(intptr_t)size);
}

int main(void)
{
    lv_init();

    /*Linux display device init*/
    lv_linux_disp_init();

    lv_obj_set_style_bg_color(lv_screen_active(), lv_color_hex(0x000000), LV_PART_MAIN);
    lv_obj_set_style_bg_opa(lv_screen_active(), LV_OPA_COVER, LV_PART_MAIN);

    lv_obj_t * parent = lv_obj_create(lv_screen_active());
    lv_obj_set_size(parent, lv_pct(50), lv_pct(50));
    lv_obj_set_style_bg_color(parent, lv_color_hex(0x191919), LV_PART_MAIN);
    lv_obj_set_style_bg_opa(parent, LV_OPA_COVER, LV_PART_MAIN);
    lv_obj_center(parent);

    lv_obj_t * layer = lv_obj_create(parent);
    lv_obj_set_size(layer, lv_pct(100), lv_pct(100));
    lv_obj_set_style_bg_opa(layer, LV_OPA_TRANSP, LV_PART_MAIN);

    //  this line introduce an error. When I comment out the line, it appears normal.
    lv_obj_update_layout(layer);

    lv_obj_t * contparent = lv_obj_create(parent);
    lv_obj_set_size(contparent, lv_pct(100), lv_pct(100));
    lv_obj_set_style_bg_opa(contparent, LV_OPA_TRANSP, LV_PART_MAIN);
    lv_obj_t * cont = lv_obj_create(contparent);
    lv_obj_set_size(cont, lv_pct(50), lv_pct(50));
    lv_obj_align(cont, LV_ALIGN_LEFT_MID, -20, 0);

    lv_obj_set_overflow_visible_flag(layer, 20);
    lv_obj_set_overflow_visible_flag(contparent, 20);
    lv_obj_set_overflow_visible_flag(parent, 20);

    /*Handle LVGL tasks*/
    while(1) {
        lv_timer_handler();
        usleep(5000);
    }

    return 0;
}
```

KioZhang, 2025-06-06:
When I adjusted the `lv_obj_update_layout` call order, it will be ok
``` c
int main(void)
{
    lv_init();

    /*Linux display device init*/
    lv_linux_disp_init();

    lv_obj_set_style_bg_color(lv_screen_active(), lv_color_hex(0x000000), LV_PART_MAIN);
    lv_obj_set_style_bg_opa(lv_screen_active(), LV_OPA_COVER, LV_PART_MAIN);

    lv_obj_t * parent = lv_obj_create(lv_screen_active());
    lv_obj_set_size(parent, lv_pct(50), lv_pct(50));
    lv_obj_set_style_bg_color(parent, lv_color_hex(0x191919), LV_PART_MAIN);
    lv_obj_set_style_bg_opa(parent, LV_OPA_COVER, LV_PART_MAIN);
    lv_obj_center(parent);

    lv_obj_t * layer = lv_obj_create(parent);
    lv_obj_set_size(layer, lv_pct(100), lv_pct(100));
    lv_obj_set_style_bg_opa(layer, LV_OPA_TRANSP, LV_PART_MAIN);

    //  this line introduce an error. When I comment out the line, it appears normal.
    // lv_obj_update_layout(layer);

    lv_obj_t * contparent = lv_obj_create(parent);
    lv_obj_set_size(contparent, lv_pct(100), lv_pct(100));
    lv_obj_set_style_bg_opa(contparent, LV_OPA_TRANSP, LV_PART_MAIN);
    lv_obj_t * cont = lv_obj_create(contparent);
    lv_obj_set_size(cont, lv_pct(50), lv_pct(50));
    lv_obj_align(cont, LV_ALIGN_LEFT_MID, -20, 0);

    lv_obj_set_overflow_visible_flag(layer, 20);
    lv_obj_set_overflow_visible_flag(contparent, 20);
    lv_obj_set_overflow_visible_flag(parent, 20);

    lv_obj_update_layout(layer);

    /*Handle LVGL tasks*/
    while(1) {
        lv_timer_handler();
        usleep(5000);
    }

    return 0;
}
```

closed 2025-06-29 by lvgl-bot

kisvegabor, 2025-06-30:
Sorry for the late reply. 

Unfortunately I can't reproduce the issue. It looks like this not matter where I place `lv_obj_update_layout` or comment it out.

[image]

KioZhang, 2025-06-30:
> Sorry for the late reply.
> 
> Unfortunately I can't reproduce the issue. It looks like this not matter where I place `lv_obj_update_layout` or comment it out.
> 
> [image]

I think the left side of the inner white object should be able to be displayed beyond the parent object, rather than truncated by the parent object
just like this green rectangle

[image]