lvgl/lvgl #8390: Crash with the Linux framebuffer and LV_DISPLAY_ROTATION_90

H5117, 2025-06-07:
### LVGL version

v9.2.2

### Platform

Linux x86_64 with gcc 14.2.1 and framebuffer backend.

### What happened?

When I add `lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90)` to my program, it crashes. With other rotation constants it works.

The reproducer is simple (resolution is 1280x800):
```
#include "lvgl/lvgl.h"

#include <unistd.h>

int main() {
    lv_init();

    lv_display_t *disp = lv_linux_fbdev_create();
    lv_linux_fbdev_set_file(disp, "/dev/fb0");
    lv_linux_fbdev_set_force_refresh(disp, true);
    
    lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90);

    lv_display_set_resolution(disp, 300, 600);
    lv_display_set_offset(disp, 300, 0);

    while(1) {
        uint32_t time_till_next = lv_timer_handler();
        usleep(time_till_next * 1000);
    }

    return 0;
}
```

### How to reproduce?

[reproducer.zip](https://github.com/user-attachments/files/20639668/reproducer.zip)

```
# gdb ./main
GNU gdb (GDB) 15.2
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./main...
(gdb) r
Starting program: /tmp/reproducer/build/main 

This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.archlinux.org>
Enable debuginfod for this session? (y or [n]) n
Debuginfod has been disabled.
To make this setting permanent, add 'set debuginfod enabled off' to .gdbinit.
[Thread debugging using libthread_db enabled]
[... 145 lines cut ...]
[Trace]	(998163.535, +0)	 call_flush_cb: Calling flush_cb on (0;0)(599;127) area with 0x7ffff796c010 image pointer lv_refr.c:1194

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7f2b22d in ?? () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff7f2b22d in ?? () from /usr/lib/libc.so.6
#1  0x00005555555a6a48 in lv_memcpy (dst=0x8000f7841000, src=0x7ffff7920010, len=512) at ../lvgl/src/stdlib/clib/lv_string_clib.c:40
#2  0x0000555555589ff6 in flush_cb (disp=0x5555555c39e0, area=0x7fffffffe3c0, color_p=0x7ffff7920010 "")
    at ../lvgl/src/drivers/display/fb/lv_linux_fbdev.c:335
#3  0x000055555556c798 in call_flush_cb (disp=0x5555555c39e0, area=0x5555555c3d6c, px_map=0x7ffff796c010 "") at ../lvgl/src/core/lv_refr.c:1211
#4  0x000055555556c645 in draw_buf_flush (disp=0x5555555c39e0) at ../lvgl/src/core/lv_refr.c:1178
#5  0x000055555556b94b in refr_area_part (layer=0x5555555c3da0) at ../lvgl/src/core/lv_refr.c:760
#6  0x000055555556b5a7 in refr_area (area_p=0x5555555c3a40) at ../lvgl/src/core/lv_refr.c:679
#7  0x000055555556b1a4 in refr_invalid_areas () at ../lvgl/src/core/lv_refr.c:586
#8  0x000055555556a9de in lv_display_refr_timer (tmr=0x5555555c3e10) at ../lvgl/src/core/lv_refr.c:398
#9  0x00005555555a61a9 in lv_timer_exec (timer=0x5555555c3e10) at ../lvgl/src/misc/lv_timer.c:326
#10 0x00005555555a5886 in lv_timer_handler () at ../lvgl/src/misc/lv_timer.c:107
#11 0x0000555555556338 in main () at ../main.c:18
(gdb) quit
A debugging session is active.

	Inferior 1 [process 194211] will be killed.

Quit anyway? (y or n) y
```

H5117, 2025-06-13:
Reproduced with LVGL 9.3.0.

AndreCostaaa, 2025-09-09:
Oh really? That's weird

What render mode are you using?

AndreCostaaa, 2025-09-18:
Hi, sorry for delay. Could you help me by logging the area before it's rotated just to see their values before the overflow?

H5117, 2025-09-18:
Hi, I used `LV_LOG_INFO("area->x1 = %i", area->x1)` and similar to print values.

At the start of `lv_display_rotate_area()`:

```
[Info]	(1232.202, +0)	 lv_display_rotate_area: area->x1 = 300 lv_display.c:1146
[Info]	(1232.202, +0)	 lv_display_rotate_area: area->y1 = 0 lv_display.c:1147
[Info]	(1232.202, +0)	 lv_display_rotate_area: area->x2 = 899 lv_display.c:1148
[Info]	(1232.202, +0)	 lv_display_rotate_area: area->y2 = 127 lv_display.c:1149
```

At the end of `lv_display_rotate_area()`:

```
[Info]	(1232.202, +0)	 lv_display_rotate_area: area->x1 = 0 lv_display.c:1181
[Info]	(1232.202, +0)	 lv_display_rotate_area: area->y1 = -300 lv_display.c:1182
[Info]	(1232.202, +0)	 lv_display_rotate_area: area->x2 = 127 lv_display.c:1183
[Info]	(1232.202, +0)	 lv_display_rotate_area: area->y2 = 299 lv_display.c:1184
```

PR #8914 (2025-09-25): fix(fbdev): fix stride, bounds checking, and partial flushes in direct render mode in flush callback
Fixes #8390 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kco [... cut]
closed 2025-09-25 by kisvegabor