lvgl/lvgl #8396: TSC Images - how to generate?

KlemenDEV, 2025-06-09:
### LVGL version

v9.3.0

### Platform

LVGL 9.3.0 with STM32U5xx with NemaGFX

### What happened?

Starting LVGL 9.3.0, NemaGFX supports TSC images - https://docs.lvgl.io/master/details/integration/renderers/nema_gfx.html#tsc-images

There is no documentation on how to generate said TSC images. Is there any publicly known information on how one would convert the images?

KlemenDEV, 2025-06-25:
Not stale, still trying to figure out where any tool for TSC images exists. Seems like library supports that format, but no docs on how to use it at this time.

KlemenDEV, 2025-06-27:
We have reached out to ST but no success so far, will keep you updated here if anything changes

AndreCostaaa, 2025-07-25:
Closing now since it seems it has been resolved. Thank you everyone :)

closed 2025-07-25 by AndreCostaaa

kisvegabor, 2025-07-28:
My pleasure! 😊