lvgl/lvgl #8397: Dark themes background show ghosting/transparency

FabioRoss, 2025-06-10:
### LVGL version

9.3.0 and earlier 9.x versions

### Platform

ESP32 S3 with OPI PSRAM
Arduino IDE & Platform.io (happens regardless)

### What happened?

When setting a dark background color, if the color is not exactly pure black, the background will show ghosting/transparency of other UI elements/screens that shouldn't be seen.

This happens also when i delete unused screens, they will still show lightly in the background.
Using lv_obj_set_style_bg_color() with lv_color_hex() or setting the background via styles (even default style) doesn't affect the result.

I'm not sure whether this is a library behavior or related to the board, given that is a RGB screen with PSRAM

[image]

[image]

[image]

vwheeler63, 2025-06-25:
I used default theme (MODE_DARK == 1) and did not see this on my hardware.  @liamHowatt , have you encountered anything like this?

vwheeler63, 2025-07-12:
Given that LVGL isn't doing that on other platforms, the cause should be relatively easy to isolate.  Using the same routine you use to get LVGL to push pixels to the display, can you manually (not using LVGL) draw a single red dot in the center that is not similarly shadowed?

FabioRoss, 2025-07-14:
> What do you mean by not "pure black"? What color depth are you using? 

Hi @AndreCostaaa, I'm using default color depth, with pure black I meant lv_color_hex(0x000000) or lv_color_black()

If I use lv_color_hex(0x111111) or any other shade of grey it will become see-through

AndreCostaaa, 2025-07-14:
Have you checked the color formats? Which one are you using and what does your screen support?

kisvegabor, 2025-07-28:
It should be a display or driver issue. Please try
- lowering or increasing the pixel clock of the display 
- leave it in the same screen for a night

Do these change anything?