lvgl/lvgl #8404: Compilation error in svg parser

wijnsema, 2025-06-10:
### LVGL version

v9.3.0

### Platform

ESP-IDF 5.4.1
Visual Studio Code, ESP-IDF plugin 1.10.0
Hardware ESP32S3 Dev module

### What happened?

With SVG option enabled,, a number of compile errors appear

### How to reproduce?

For example: 
`In function '_skip_space_and_separators':   
C:/Projects/button30/managed_components/lvgl__lvgl/src/libs/svg/lv_svg_parser.c:438:39: error: array subscript has type 'cha 
' [-Werror=char-subscripts]
  438 |     while((str < str_end) && (isspace(*str) || _is_separators(*str))) {`

I was able to fix this by casting the `*str` to `(unsigned char)` in the call to `isspace`, but I'm sure there are better solutions.

wijnsema, 2025-06-25:
Still an issue

kisvegabor, 2025-06-27:
@AndreCostaaa can you take a look at it?

AndreCostaaa, 2025-07-03:
Hi @wijnsema 

Sorry for taking so much time to get back to you. I'm not able to reproduce your problem, which compiler and version are you using?

wijnsema, 2025-07-04:
Steps to reproduce:

- Use ESP-IDF extension (version 1.10.2) in Visual Studio Code, in my case on Windows 11
- Install ESP-IDF  v5.4.2 (under Advanced section in extension) if not already installed
- Create a New Project with ESP-IDF V5.4.2 ->  Target IDF_TARGET esp32S3, choose hello_world template
- Open Terminal Shell and add LVGL like this: ```idf.py add-dependency "lvgl/lvgl^9.3.0"```
- Add ```REQUIRES lvgl``` to the CMakelist.txt file of the hello_world project
- Now ```idf.py menuconfig``` (maybe build project first, not sure if needed)
- In menuconfig change ```LV_VECTOR_GRAPHICS``` to yes
- In menuconfig change ```LV_USE_SVG```
- Save the configuration and start the build

Boom, next  errors occur:
```
In file included from C:/Projects/lvgl542/managed_components/lvgl__lvgl/src/libs/svg/lv_svg_parser.c:13:
C:/Projects/lvgl542/managed_components/lvgl__lvgl/src/libs/svg/lv_svg_parser.c: In function '_skip_space':
C:/Projects/lvgl542/managed_components/lvgl__lvgl/src/libs/svg/lv_svg_parser.c:425:38: error: array subscript has type 'char' [-Werror=char-subscripts]
  425 |     while((str < str_end) && isspace(*str)) {
      |                                      ^~~~
C:/Projects/lvgl542/managed_components/lvgl__lvgl/src/libs/svg/lv_svg_parser.c: In function '_skip_space_and_separators':  
C:/Projects/lvgl542/managed_components/lvgl__lvgl/src/libs/svg/lv_svg_parser.c:438:39: error: array subscript has type 'char' [-Werror=char-subscripts]
  438 |     while((str < str_end) && (isspace(*str) || _is_separators(*str))) {
      |                                       ^~~~
C:/Projects/lvgl542/managed_components/lvgl__lvgl/src/libs/svg/lv_svg_parser.c: In function '_parse_color':
C:/Projects/lvgl542/managed_components/lvgl__lvgl/src/libs/svg/lv_svg_parser.c:518:28: error: array subscript has type 'char' [-Werror=char-subscripts]
  518 |             if(isxdigit(str[1]) && isxdigit(str[2]) && isxdigit(str[3])) {
      |                         ~~~^~~
C:/Projects/lvgl542/managed_components/lvgl__lvgl/src/libs/svg/lv_svg_parser.c:518:48: error: array subscript has type 'char' [-Werror=char-subscripts]
  518 |             if(isxdigit(str[1]) && isxdigit(str[2]) && isxdigit(str[3])) {
      |                                             ~~~^~~
C:/Projects/lvgl542/managed_components/lvgl__lvgl/src/libs/svg/lv_svg_parser.c:518:68: error: array subscript has type 'char' [-Werror=char-subscripts]
  518 |             if(isxdigit(str[1]) && isxdigit(str[2]) && isxdigit(str[3])) {
      |                                                                 ~~~^~~
C:/Projects/lvgl542/managed_components/lvgl__lvgl/src/libs/svg/lv_svg_parser.c:529:28: error: array subscript has type 'char' [-Werror=char-subscripts]
  529 |             if(isxdigit(str[1]) && isxdigit(str[2]) && isxdigit(str[3])
      |                         ~~~^~~
C:/Projects/lvgl542/managed_components/lvgl__lvgl/src/libs/svg/lv_svg_parser.c:529:48: error: array subscript has type 'char' [-Werror=char-subscripts]
  529 |             if(isxdigit(str[1]) && isxdigit(str[2]) && isxdigit(str[3])
      |                                             ~~~^~~```

etc...

As stated, easy to fix with a cast, but these are not my files.

wijnsema, 2025-07-22:
Still an issue

AndreCostaaa, 2025-07-25:
PR open. Sorry for the delay and thank you for opening this issue

PR #8619 (2025-07-30): fix(svg): cast char to unsigned preventing char subscription errors
Fixes #8404 

Error occurs using the xtensa compiler

Also had to fix a thorvg warning to get the application to compile so putting it here as well

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/mas [... cut]
closed 2025-07-30 by liamHowatt