lvgl/lvgl #8416: Drawing lines outside of a container object

lorenzo-arena, 2025-06-11:
### LVGL version

v9.3.0

### Platform

Custom board with a STM32H753 MCU, also simulated on Linux using SDL

### What happened?

I'd like to draw a line outside of its container object. I want to draw it "on the left" of the container, with points with X coordinates going from 0 to negative values. I've tried the modifying the line example but without success:

[image]

### How to reproduce?

This is a modified version of lv_example_line_1.c; I've tried it in the lv_port_vs_code repository.

```c
#include "../../lv_examples.h"
#if LV_USE_LINE && LV_BUILD_EXAMPLES

static void ext_draw_size_event_cb(lv_event_t *e)
{
    lv_event_set_ext_draw_size(e, 30);
}

void lv_example_line_1(void)
{
    /*Create an array for the points of the line*/
    // static lv_point_precise_t line_points[] = { {5, 5}, {70, 70}, {120, 10}, {180, 60}, {240, 10} };
    static lv_point_precise_t line_points[] = { {0, 0}, {-30, 0}, {-30, 220}, {-15, 220}, {0, 220} };

    /*Create style*/
    static lv_style_t style_line;
    lv_style_init(&style_line);
    lv_style_set_line_width(&style_line, 8);
    lv_style_set_line_color(&style_line, lv_palette_main(LV_PALETTE_BLUE));
    lv_style_set_line_rounded(&style_line, true);

    /*Create a container*/
    lv_obj_t * container;
    container = lv_obj_create(lv_screen_active());
    lv_obj_remove_style(container, NULL, LV_PART_MAIN);
    lv_obj_set_scrollbar_mode(container, LV_SCROLLBAR_MODE_OFF);
    lv_obj_set_size(container, 200, 200);
    lv_obj_align(container, LV_ALIGN_CENTER, 0, 0);
    lv_obj_set_style_bg_opa(container, LV_OPA_COVER, LV_PART_MAIN);
    lv_obj_set_style_bg_color(container, lv_color_hex(0xAAAAAA), LV_PART_MAIN);

    lv_obj_add_flag(container, LV_OBJ_FLAG_OVERFLOW_VISIBLE);
    lv_obj_add_event_cb(container, ext_draw_size_event_cb, LV_EVENT_REFR_EXT_DRAW_SIZE, NULL);

    /*Create a line and apply the new style*/
    lv_obj_t * line1;
    line1 = lv_line_create(container);
    lv_line_set_points(line1, line_points, 5);     /*Set the points*/
    lv_obj_add_style(line1, &style_line, 0);
}

#endif
```

krembed, 2025-06-11:
Shouldn't `LV_EVENT_REFR_EXT_DRAW_SIZE` be used on the line and not the container?

krembed, 2025-06-11:
What happens when instead of using negative values directly you move the line with `lv_obj_set_pos(line1,....)`? I can't really test it myself right now I apologize.

lorenzo-arena, 2025-06-11:
That seems to make it work! If I modify the `line_points` array from 

```c
static lv_point_precise_t line_points[] = { {0, 0}, {-20, 0}, {-20, 220}, {-10, 220}, {0, 220} };
```

to

```c
static lv_point_precise_t line_points[] = { {20, 0}, {0, 0}, {0, 220}, {10, 220}, {20, 220} };
```

and after creating the line I also set its position with

```c
lv_obj_set_pos(line1, -20, 0);
```

the following is the result:

[image]

RonTang, 2025-06-13:
I have the same problem with version 9.3.0. I can't use negative coordinates to draw a line.

lorenzo-arena, 2025-06-13:
I also tried bisecting a little bit but the issue is there at least from v9.0.0

lorenzo-arena, 2025-07-16:
I'll close this one since it's expected for lines to "not work" with negative coordinates if the extra draw size is not set both on the line and on the container if necessary.

closed 2025-07-16 by lorenzo-arena