lvgl/lvgl #8417: Long mode dots or clip does work

LuisCRSousa, 2025-06-11:
### LVGL version

v9.3.0

### Platform

STM32 and simulator with wsl

### What happened?

LV_LABEL_LONG_MODE_DOTS and LV_LABEL_LONG_MODE_CLIP does not work with v9.3.0. It works with v9.2.2.
Instead of dots or clip it is always wrapping it to the next lines. Circular and scroll works.

### How to reproduce?

```
lv_obj_t* label = lv_label_create(get_lv_obj());
lv_obj_set_width(label, 100);
lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
lv_label_set_text(label, "this is a long text that should be clipped with dots");
lv_label_set_long_mode(label, LV_LABEL_LONG_MODE_DOTS);
```

yihewodeyihe, 2025-06-11:
Try to limite the size of the area, not only the width. The default height of label may be LV_SIZE_CONTENT.

LuisCRSousa, 2025-06-11:
Setting the height to 20 made it work. But why? If i'm setting a fixed width shouldn't it work independently of the height?

kisvegabor, 2025-06-26:
Hey,

I confirm that what you see is the intended behavior. 

In `DOTS` mode with fixed with the text is word wrapped, if height is `LV_SIZE_CONTENT` the label size will be set as needed.

Dots will be added only if height fixed and not large enough. This way you can create multi lines texts with dots at the end. E.g. by adding `lv_obj_set_height(label, 50);`

[image]

I'm thinking for long about adding a speciel unit for line count. E.g. `lv_obj_set_height(label, LV_EM(3));` (3 lines of text)

I think it'd make it more intuitive.