lvgl/lvgl #8420: Tab View does not send a LV_EVENT_VALUE_CHANGED event, despite the documentation saying it does

RLH-2110, 2025-06-11:
### LVGL version

9.3.0

### Platform

STM32MPU135F-DK with OpenSTLinux
LVGL with SDL2

### What happened?

[The documentation](https://docs.lvgl.io/9.3/details/widgets/tabview.html#events) says that:
> [LV_EVENT_VALUE_CHANGED](https://docs.lvgl.io/9.3/API/misc/lv_event.html#_CPPv4N15lv_event_code_t22LV_EVENT_VALUE_CHANGEDE) Sent when a new tab is selected by sliding or clicking the tab button.

But that event never seems to occur when a tab is pressed

### How to reproduce?

Here is a program that I would expect to work. It works if I use LV_EVENT_ALL instead of LV_EVENT_VALUE_CHANGED.

```C
#include <stdint.h>
#include <pthread.h>
#include <unistd.h>
#include <signal.h>

#include "lvgl/lvgl.h"
#include "lvgl/src/drivers/sdl/lv_sdl_mouse.h"

#define H_RES (480)
#define V_RES (272)

#define LVGL_DESIRED_TICK_INCREASE_MS 5

#define MICROSECOND_TO_MILISECOND_RATE 1000

/* lvgl stuff and threats */
static lv_display_t *disp;
static lv_indev_t *lvMouse;

static pthread_t tickThread;
static void* tick_thread(void* data);
volatile sig_atomic_t stop = 0; /* programms runs as long as this is set */

/* callbacks */
void tab_view_value_changed_cb(lv_event_t *e);
uint32_t u32Temp;

int main(){

  lv_init();

  pthread_create( &tickThread, NULL, tick_thread,NULL);

  disp = lv_sdl_window_create(H_RES, V_RES);
  lvMouse = lv_sdl_mouse_create();

  lv_obj_t *screen = lv_obj_create(NULL);

  lv_obj_t *tabView = lv_tabview_create(screen);

  lv_obj_t *tabHome = lv_tabview_add_tab(tabView, "Home");
  lv_obj_t *tabExit = lv_tabview_add_tab(tabView, "Exit");

  u32Temp = lv_obj_get_index(tabExit);
  lv_obj_add_event_cb(tabView,tab_view_value_changed_cb,LV_EVENT_VALUE_CHANGED, &u32Temp);

  lv_screen_load_anim(screen,LV_SCR_LOAD_ANIM_NONE,0,0,true);

  while(!stop){
    uint32_t timeTillNext = lv_timer_handler();
    usleep(timeTillNext*MICROSECOND_TO_MILISECOND_RATE );
  }

  pthread_join(tickThread,NULL);
  lv_indev_delete(lvMouse);
  lv_disp_remove(disp);
  lv_sdl_quit();
  lv_deinit();
}

/* callbacks */
void tab_view_value_changed_cb(lv_event_t *e){
  lv_obj_t *tabView = lv_event_get_target_obj(e);
  int32_t exitTabId = *((int32_t*)lv_event_get_user_data(e));

  if (lv_tabview_get_tab_active(tabView) == exitTabId)
    stop = 1;
}

/* other functions */
static void *tick_thread(void* data) {
  (void) data;
  while(!stop) {
    usleep(LVGL_DESIRED_TICK_INCREASE_MS * MICROSECOND_TO_MILISECOND_RATE);
    lv_tick_inc(LVGL_DESIRED_TICK_INCREASE_MS);
  }
}
```

AdrianGITHUB2015, 2025-06-20:
lvgl9.3.0版本，我也遇到过这个问题。我不清楚是不是设置问题 或者是 tabview 控件注册事件的方式不对；
但我通过检查 lv_tabview.c文件源码发现，在 lv_tabview_set_active 函数中 tabview->tab_cur 被过早的赋值，导致在 cont_scroll_end_event_cb 回调函数中计算的 局部变量t 与  vlv_tabview_get_tab_active(tv) 的值相等 这被认为tab 没有更改， 所以 if(new_tab) lv_obj_send_event(tv, LV_EVENT_VALUE_CHANGED, NULL);  语句中 new_tab == false 没有向 tv 发送 LV_EVENT_VALUE_CHANGED 事件。
我的解决方案是 参考了 之前的lvgl版本代码， 修改 lv_tabview_set_active 函数 将 tabview->tab_cur = idx; 语句放到了 函数的末尾，程序正常运行。

liamHowatt, 2025-08-04:
Closing as it should be fixed by #8537

closed 2025-08-04 by liamHowatt