lvgl/lvgl #8435: Active screen doesn't receive input from indev, only first active does

jeanthom, 2025-06-16:
### LVGL version

v9.3.0

### Platform

RP2040 with NuttX 12.9.0.

### What happened?

1. Create a handleful `lv_obj_create(NULL)`
2. Populate them with widgets
3. Start the app with the first created screen active
4. Set another screen active (`lv_async_call((lv_async_cb_t)lv_screen_load, my_other_screen);`)
5. Notice that the screen activated in 4 doesn't respond to input
6. Go back to the screen active in 3
7. Notice that while the screen in 4 was "active", the first created screen was capturing all input.

brightproject, 2025-06-16:
> 4\. Set another screen active (`lv_async_call((lv_async_cb_t)lv_screen_load, my_other_screen);`)

Try another way🙂
    
```
// lv_obj_t * scale_screen = lv_obj_create(NULL);
scale_screen = lv_obj_create(NULL);
lv_obj_clear_flag(scale_screen, LV_OBJ_FLAG_SCROLLABLE);      /// Flags
lv_obj_set_pos(scale_screen, 0, 0);
lv_obj_set_size(scale_screen, 480, 480);
```

brightproject, 2025-06-17:
> Could you explain in details what this is supposed to do?

This is the answer to your question.
When you specify `NULL`, the library assumes that all your screens are first by default.

> https://docs.lvgl.io/master/details/main-modules/display/setup.html#default-display

jeanthom, 2025-06-17:
> This is the answer to your question. When you specify `NULL`, the library assumes that all your screens are first by default.
> 
> > https://docs.lvgl.io/master/details/main-modules/display/setup.html#default-display

I'm confused. I only have one display configured. Maybe you are mixing screens and displays? In LVGL terminology those describe two distinct things: https://docs.lvgl.io/master/details/common-widget-features/basics.html#what-are-screens

Either way I'm still trying to figure a way to get only the active screen to receive inputs from the default group.

jeanthom, 2025-06-18:
I found a workaround: register a "screen loaded" event handler and make it focus group to the desired widget. This is actually something quite sensible to do.

I still can't figure out why it is now needed.

jeanthom, 2025-07-20:
Stull active...

kisvegabor, 2025-09-27:
Hi,

It might be too late, but I confirm that the focused widget can be on a not loaded screen too, which is really silly. 
We need to improve it for sure.