lvgl/lvgl #8444: v9.4 Planning

kisvegabor, 2025-06-17:
### Introduce the problem

For this release let's set up clear schedule upfront and execute feature freeze and the release accordingly:
- September 15, 2025: Start the feature freeze period (merge only critical fixes)
- October 1, 2025: Release v9.4

---

Planned features:
- [ ] OpenGL 3D renderer (gltf) support
- [ ] Finalize XML support
- [ ] Improve the docs (add description for caching, vector API, etc)

---

Please comment about any ideas that you would like to see in LVGL v9.4

wijnsema, 2025-06-17:
More stable SVG support, as SVG in 9.3 it is hardly working. And seems to be low priority.

SVG could be a killer feature.

wijnsema, 2025-06-18:
> What is the main issue that you see?

I cannot get it to work at all actually, see #8405 

@itavero got it working but rendering is not correct,

Also, compiler flags needed are not enforced/documented.

brightproject, 2025-06-18:
There will be a clear integration of `LVGL` and `freeRTOS`, with examples for versions 9.3 and 9.4

vwheeler63, 2025-06-18:
If there are any outstanding bugs, I would think these should be a priority so new projects can feel safe to start with 9.3/9.4....

jeanthom, 2025-06-19:
@kisvegabor I created https://github.com/lvgl/lvgl/issues/8455 to track the menu navigation issue

onecoolx, 2025-06-20:
@kisvegabor  Do we have plans to support html widget? It is more friendly to large amounts of text layout rendering.

alexriegler12, 2025-06-20:
Make the scale label gap configurable. Currently, it is a hardcoded constant.

onecoolx, 2025-06-22:
> [@onecoolx](https://github.com/onecoolx) I agree that we should achieve large amount of clean text rendering, it's a very good idea. However HTML - I am not a big fan of. IMHO it's way too complex for LVGL which is supposed to be **light** and **versatile** :) HTML is not light nor versatile. I believe we should go 'old school' on this and use SGML instead. https://www.tei-c.org/Vault/GL/P3/SG.htm no attributes - no headaches. XML/HTML is LR+4. if we only need to render text, so using only entities things like h1 h2, b should suffice and if we can apply styles to each entity it would be already very useful and a cool feature.
> 
> Are we building a web browser out of LVGL :) ? I hope not...
> 
> To achieve good text rendering we may have to consider to use Harfbuzz for text shaping, we also need to figure out how to properly scale Freetype outlines. Retrieving the outlines in font units and then scaling and doing grid-fitting properly not 128px. The current Freetype integration is a very good start but let's test it with more fonts and more sizes. It requires time to really understand digital typography...

Yes, you are right. A full-featured browser is a huge and complex project. I have worked on it for many years and know how complex it is. But we may not need a full-featured browser, we just need to be able to do text layout in a simple way. There are some very small, lightweight browsers that may achieve this goal, such as "dillo".

https://dillo.org/
https://github.com/dillo-browser/dillo

yushuailong, 2025-06-24:
Text rendering and update_layout are very time-consuming tasks for me. It would be great if we could improve them in V9.4. :smile:

AndreCostaaa, 2025-06-29:
Complete dmabuf support for wayland

IAMMX, 2025-07-04:
Can the lv_table be modified to add the functionality of freezing the first row？

Since the table is quite long, when scrolling to the bottom, the header row (usually containing titles) becomes hidden, making it difficult to view intuitively. Therefore, we request the addition of a header row freezing feature.

IAMMX, 2025-07-11:
> The same effect be realized if you add a simple one-row-table on to of the real one. Could this trick work for you?

I think it would be best to implement this feature by adding an API, such as `lv_table_freeze_first_row`.

lizhaoming634, 2025-07-11:
> How to achieve this chart effect in LVGL 9? Could you provide an example?如何在 LVGL 9 中实现这种图表效果？您能举个例子吗？
> 
> [image]

@kisvegabor  Could you provide the updated code for implementing the effects currently demonstrated on the Demos page

lizhaoming634, 2025-07-14:
> Hi [@lizhaoming634](https://github.com/lizhaoming634) The widgets demo code is available here https://github.com/lvgl/lvgl/tree/master/demos/widgets

[image]
Thanks for the demo link! I've tried the latest widgets demo code, but my output doesn't look identical to the web version. Could you please take a look? Left side is my rendering, right side is the expected web result

kisvegabor, 2025-07-14:
@lizhaoming634 just added stacked chart support here: https://github.com/lvgl/lvgl/pull/8573

IAMMX, 2025-07-29:
Is there a plan to support SDL3?

Viatorus, 2025-08-11:
Because you are supporting Figma in LVGL UI Editor: https://github.com/lvgl/lvgl/issues/6131

psi-sue, 2025-08-25:
> > What is the planned way of interacting with C arrays of strings (or structs) from XML? Subjects can only handle simple types like ints and strings, but not more complex types like arrays.
> 
> Great question. That's also something that I'm actively thinking about. Just to make sure we mean the same thing. For example you have a `week_day_int` subject (values are from 0-6) and and array with week day names ("Monday", "Tuesday", ...). The questions is how to map the string on label depending on `week_day_int`. The options I see:
> 
>     1. Create an other subject called `week_day_name` write a simple C code in `ui.c` that adds an observer to `week_day_int` and set the value of `week_day_name` when `week_day_int` changes.
> 
>     2. We do that same in XML, e.g.
> 
> 
> <subject_string_array_mapper src="week_day_int" dest="week_day_name" array="'Monday', 'Tuesday', ..."/> 
> 
> What do you think?

I was thinking more about a simple todo app, where a user could add a todo (with some text and status), then display this list on the UI. Of course the array will be dynamic in this case and all elements have to be displayed and the status of the todo would also have to be taken into account (so maybe different style depending on the state of the todo)

Something like:

```
typedef struct {
    char* text;
    bool status;
} Todo;

typedef struct {
    Todo* items; 
    size_t size;
    size_t capacity;
} TodoList;
```

Or is this too custom and requires a new widget?

Satnet, 2025-08-25:
> [@Satnet](https://github.com/Satnet)
> 
> > Will LVGL 9.4 have multi touch - as in being able to click two different things with two different fingers?
> 
> It's already supported. Just create 2 indevs, the first ready finger 0, the other reads finger 1.

Thank you. I tried it and trying to move two sliders at the same time will cause none of them to move.
I guess that might work when gesture support is implemented?

amroyat23-prog, 2025-09-22:
Will more effects like blur and more detailed and beautiful charts be added in the next version? Something close to QT embedded

closed 2025-10-16 by AndreCostaaa