lvgl/lvgl #8449: v9.3 doesn't work on ESP32S3 with ili9341 displays

dariomingarelli, 2025-06-18:
### LVGL version

9.3

### Platform

MCU: ESP32-S3-WROOM-1-N8R8
DISPLAY CONTROLLER: ILI9341
lvgl/lvgl: "9.3"
espressif/esp_lcd_ili9341: "2.0.0"
idf: "5.4.1"

### What happened?

A project perfectly working with LVGL 9.2.0 doesn't work with 9.3.

The app run, but nothing is drawn on the display.

I've changed configuration as described on "[LVGL doesn't start, randomly crashes or nothing is drawn on the display. What can be the problem?](https://docs.lvgl.io/master/intro/introduction.html#lvgl-doesn-t-start-randomly-crashes-or-nothing-is-drawn-on-the-display-what-can-be-the-problem)", but still nothing drawn.

How can I do to run also 9.3, instead of 9.2?

vwheeler63, 2025-06-25:
@liamHowatt Do you know of any v9.2 -> v9.3 changes that might be connected to this?

liamHowatt, 2025-07-03:
Nothing really changed with mipi and ili9341 since 9.2.0 was tagged August 27, 2024.

https://github.com/lvgl/lvgl/commits/master/src/drivers/display
https://github.com/lvgl/lvgl/tags

It could be something else.

Are you able to share more info, i.e. your code, config, etc.?

dariomingarelli, 2025-07-03:
You can experience that in this way:
- Use ESP-IDF 5.4.2 or 5.4.1
- Use Espressif-IDE 3.5.0
- Start new Espressif IDF Project
- Select ESP32S3
- Select "Create a project using one of the templates"
- Choose "peripherals > lcd > spi_lcd_touch"

Build and debug... EVERYTHING WORKS!
 
- Open "main > idf_component.yml" file and change from:
dependencies:
  lvgl/lvgl: "9.2.0"
  esp_lcd_ili9341: "^1.0"
  esp_lcd_gc9a01: "^1.0"
  esp_lcd_touch_stmpe610: "^1.0"

To:
dependencies:
  lvgl/lvgl: "9.3.0"
  esp_lcd_ili9341: "^2.0"
  esp_lcd_gc9a01: "^1.0"
  esp_lcd_touch_stmpe610: "^1.0"

Build and debug... NOTHING APPEAR ON THE DISPLAY, without any error in log.
Same issues also using:
  lvgl/lvgl: "9.2.1"
  esp_lcd_ili9341: "^1.0"

  lvgl/lvgl: "9.2.2"
  esp_lcd_ili9341: "^1.0"

  lvgl/lvgl: "9.2.1"
  esp_lcd_ili9341: "^2.0"

  lvgl/lvgl: "9.2.2"
  esp_lcd_ili9341: "^2.0"

AndreCostaaa, 2025-07-03:
Could it be because of the update to this library `esp_lcd_ili9341: "^2.0"` and unrelated to LVGL ?

dariomingarelli, 2025-07-03:
Same issue with:
lvgl/lvgl: "9.3.0"
esp_lcd_ili9341: "^1.0"

dariomingarelli, 2025-07-03:
I think that something happen from 9.2.0 to 9.2.1...

It could be something encountered or known by Espressif as well, since they blocked all examples with LVGL at version 9.2.0.