lvgl/lvgl #8452: LV_SIZE_CONTENT doesn't take aligned objects into account

lorenzo-arena, 2025-06-18:
### LVGL version

v9.3.0

### Platform

Custom board with a STM32H753 MCU, also simulated on Linux using SDL

### What happened?

When creating a container object with size set to `LV_SIZE_CONTENT` set as the width or height, if the inner children are aligned for example using `lv_obj_align_to` the parent size isn't updated accordingly. Is this expected?

closed 2025-07-28 by lvgl-bot

kisvegabor, 2025-07-28:
Sorry for the delay. 

This is expected for multiple reasons
- align out options align the object outside of the reference objects and children out of the parent are clipped 
- LV_SIZE_CONTENT changes only the bottom or right coordinate. (Left with right-to-left base direction)
- even some inner alignments are ignored to avoid circular dependency. E.g. align to the right with 5 px offset, but increase the size to involve the children, but align the child to the right +5px...

lorenzo-arena, 2025-07-28:
Thanks! I'll work around that then, I guess just by setting explicit sizes and positions for the various widgets in those cases