lvgl/lvgl #8477: issues with g2d drawing images from LVGLs Image Converter

krembed, 2025-06-24:
### LVGL version

v9.3.0-70-geee9ffa55

### Platform

Yocto ARM64

### What happened?

There is an issue I ran into with g2d. I have linked against libg2d and converted an image using the online converter. I set the cf to ARGB8888. After declaring the image and running the app I run into:
`g2d_blit_wrap: Invalid src rect, left 0, top 0, right 225, bottom 160, width 225, height 160, stride 0!`
Image data:
`Image w=225 h=160 cf=16 data_size=144000`

It looks like the problem might be in `lv_draw_g2d_img.c` inside `lv_draw_g2d_img`
As far as I understand this line:
`const lv_image_dsc_t * img_dsc = dsc->src;`
will access the array directly? That means in this line 
`int32_t src_stride = img_dsc->header.stride / (lv_color_format_get_bpp(img_dsc->header.cf) / 8);`
src_stride ends up being 0 for me as the stride is not set in that generated file containing image data.

If I use `lv_draw_buf_width_to_stride and pass` `header.w` and `header.cf` to it instead of `img_dsc->header.stride` it seems to work fine and g2d will draw that image.

AndreCostaaa, 2025-06-24:
You're right, I was able to replicate the issue with a random image. This is a bug with the image converter as it should generate the stride

krembed, 2025-06-24:
I misunderstood the cause of the issue then, I have changed the title accordingly if that is fine

AndreCostaaa, 2025-06-24:
Yep, your analysis is correct. It's better to calculate it using `lv_draw_buf_width_to_stride` as tha. Do you want to open a PR or should I?

krembed, 2025-07-11:
not stale

AndreCostaaa, 2025-07-25:
Closing this as the issue has been fixed. Thanks !