lvgl/lvgl #8487: Commit 'decouple scroll momentum decay from indev read loop' causes undesired behaviour

pete-pjb, 2025-06-25:
### LVGL version

Master branch

### Platform

AMD Ultrascale+ MPSOC (Cortex A53 cores) using display port hardware.

### What happened?

Since this change there is no synchronisation between the movement of sliders and scrolling items and the position of your finger on the touch screen, please consider reversing this change.  Thanks.

[Here](https://github.com/lvgl/lvgl/commit/c04a508d3501dd87847118abe87a823406a982c6) is the commit concerned.

### How to reproduce?

Just move a slider or scroll a scrollable object backwards and forwards without lifting your finger from the screen and you will see the object move complete out of synchronisation with your finger movement.

AndreCostaaa, 2025-06-25:
Thank you for opening this issue @pete-pjb. The commit has only been live for a couple of hours so it's nice to have such quick feedback

I'm ccing the original author @niklasf as he's most aware of the implementation and may have a solution

niklasf, 2025-06-25:
To potentially narrow it down, which input driver in lvgl are you using on this platform and which tick source?

niklasf, 2025-06-25:
Ah. It's probably due to the new `data->timestamp` then.

> If the driver can provide precise timestamps for buffered events, it can overwrite `data->timestamp`. By default, this is initialized to `lv_tick_get()` just before invoking `read_cb`.

So effectively this happens internally:

```c
data->timestamp = lv_tick_get(); // default
mouse_read(..., data);
// data->timestamp is used
```

This looks generally backwards compatible due to the default, but it's actually a breaking change for implementations that copy or memset the entire `lv_indev_data_t`.

Can you please do

```c
data->timestamp = lv_tick_get(); 
```

as the last step in `mouse_read` to verify?

If confirmed, we have to decide what to do.

kisvegabor, 2025-06-25:
Thank you all for the quick action and debugging! 

@pete-pjb in your case `data->timestamp` was set to 0, right? Maybe we can just add a sanity check and switch the logic:
```c
data->timestamp = 0;
mouse_read(..., data);
if(data->timestamp == 0) data->timestamp = lv_tick_get(); /*default*/
```

pete-pjb, 2025-06-25:
Hi @kisvegabor ,

Hope you and family are well!

Yes in my case it would have been zero as it is stored in my global structures which are all initialised to 0 based on size alone at the beginning of runtime.

I would say your solution is good as long as the user initialises their data to zero, I expect most programmers would do so anyway, so it should deal with most cases in my opinion.

It might be worth adding something to the documentation to mention the new 'indev' timestamp feature and to warn people to initialise it to zero if they don't wish to use it... 

What do you think? 🙂 

Cheers,  

Pete

niklasf, 2025-06-25:
Some more ideas:

* Future proof `lv_indev_data_t` by making it opaque and instead provide getters and setters. Will cause some churn once.
* Revert this part of the patch. Is there a concrete driver that actually is able to provide better timestamps in a noticable way? 
* Support zero-initialization without a special `0` value, by requiring an additional `data->timestamp_provided = true` for drivers that want it.
* Make the feature a property of the input device: `lv_indev_set_provides_timestamps(lv_indev_t*, bool)` or similar.

kisvegabor, 2025-06-26:
> I would say your solution is good as long as the user initialises their data to zero, I expect most programmers would do so anyway, so it should deal with most cases in my opinion.

Not initializing to zero is considered undefined behaviour, which will really happen, so it's ok.

> It might be worth adding something to the documentation to mention the new 'indev' timestamp feature and to warn people to initialise it to zero if they don't wish to use it...

Sure, it can be mentioned [here](https://github.com/lvgl/lvgl/commit/c04a508d3501dd87847118abe87a823406a982c6#diff-0dfe6083a48780a16e8c0500df27ab7f4f68756f44fd7314b1832ce72848d816R662-R665).

---

@niklasf 
Thank you for the other suggestions. I think the lowest hanging fruit it the fix that I have proposed. If you all agree could you send a PR?

niklasf, 2025-06-30:
Beat me to it. Thanks!