lvgl/lvgl #8497: Tabview with list widget in tab fails at runtime with v.9.3

supremeneuron, 2025-06-28:
### LVGL version

9.3

### Platform

Arduino 2.3.6, ESP32 v. 3.1.3 core compiled on Windows 11 Pro
Devices tested are ESP32C3 and ESP32C6 (same issue with both devices)

### What happened?

I've run into an unusual issue.  I have a tabview control and in one tab I have a list control.  In v9.2.x, this code works perfectly but in v9.3, it compiles but at runtime a crash occurs.  Unfortunately, I can't seem to get a debug dump so I don't have any debugging information.

Below is a sample code that results in the error.  Basically, it is a combination of the tabview and list control from the website.  Nothing fancy, but does consistently reproduce the error.

### How to reproduce?

void lv_example_tabview_1(void)
{
    /*Create a Tab view object*/
    lv_obj_t * tabview;
    tabview = lv_tabview_create(lv_screen_active());

    /*Add 3 tabs (the tabs are page (lv_page) and can be scrolled*/
    lv_obj_t * tab1 = lv_tabview_add_tab(tabview, "Tab 1");
    lv_obj_t * tab2 = lv_tabview_add_tab(tabview, "Tab 2");
    lv_obj_t * tab3 = lv_tabview_add_tab(tabview, "Tab 3");

    /*Add content to the tabs*/
    lv_obj_t * label = lv_label_create(tab1);
    lv_label_set_text(label, "This the first tab\n\n"
                      "If the content\n"
                      "of a tab\n"
                      "becomes too\n"
                      "longer\n"
                      "scrollable.\n"
                      "\n"
                      "\n"
                      "\n"
                      "Can you see it?");

    label = lv_label_create(tab2);
    lv_label_set_text(label, "Second tab");

    /*Create a list*/
    list1 = lv_list_create(tab3);
    lv_obj_set_size(list1, 180, 220);
    lv_obj_center(list1);

    /*Add buttons to the list*/
    lv_obj_t * btn;
    lv_list_add_text(list1, "File");
    btn = lv_list_add_button(list1, LV_SYMBOL_FILE, "New");
    lv_obj_add_event_cb(btn, event_handler, LV_EVENT_CLICKED, NULL);
    btn = lv_list_add_button(list1, LV_SYMBOL_DIRECTORY, "Open");
    lv_obj_add_event_cb(btn, event_handler, LV_EVENT_CLICKED, NULL);
    btn = lv_list_add_button(list1, LV_SYMBOL_SAVE, "Save");
    lv_obj_add_event_cb(btn, event_handler, LV_EVENT_CLICKED, NULL);

    lv_list_add_text(list1, "Exit");
    btn = lv_list_add_button(list1, LV_SYMBOL_OK, "Apply");
    lv_obj_add_event_cb(btn, event_handler, LV_EVENT_CLICKED, NULL);
    btn = lv_list_add_button(list1, LV_SYMBOL_CLOSE, "Close");
    lv_obj_add_event_cb(btn, event_handler, LV_EVENT_CLICKED, NULL);

    lv_obj_scroll_to_view_recursive(label, LV_ANIM_ON);

}

supremeneuron, 2025-07-09:
I could not find a solution to the above situation, so I resorted to using a table widget and modified the code to render the necessary functionality with a table rather than a list within a tab.  This has worked well.

AndreCostaaa, 2025-10-17:
> Also if you could please enable logging and provid them it would be very useful to help diagnose this

Also this please

AndreCostaaa, 2025-12-10:
Hi, I've tried reproducing it and unfortunately I can't. As the issue has been stale for a long time, i'll be closing this down for now

If you still encounter it, feel free to get back to me with the application logs whenever you get the chance and i'll reopen it back, thank you!