lvgl/lvgl #8517: Wayland DMA Buffer wait for flushing

cosmindanielradu19, 2025-07-02:
### LVGL version

v9.3.0

### Platform

NXP i.MX MPU

### What happened?

When LVGL is configured to use both software and G2D rendering units, along with DMA buffers, on a 1080p HDMI-connected display, visual artifacts appear and the LVGL window flickers.

After enabling Wayland debug flags, I observed that LVGL begins buffer swapping and drawing immediately after `lv_wayland_wait_flush_cb` completes. However, the `buffer_release` callback from Wayland is triggered *after* LVGL starts drawing.

This indicates a synchronization issue: LVGL should wait for the `buffer_release` signal before acquiring a buffer for rendering. Without this wait, drawing begins when the buffer is still used by the compositor, leading to flickering and display artifacts.

I believe fixing this issue may require some changes on the `lv_refr` side, so I'd appreciate your input on how best to approach it.

@AndreCostaaa @kisvegabor do you have any ideas on how we could approach fixing this?

AndreCostaaa, 2025-07-02:
Hi @cosmindanielradu19. I didn't notice the flickering but it makes sense that it would be happening

What is stopping us from simply waiting for the `buffer_release` event before making LVGL render the next frame ?

cosmindanielradu19, 2025-07-02:
> Hi [@cosmindanielradu19](https://github.com/cosmindanielradu19). I didn't notice the flickering but it makes sense that it would be happening

i.MX 93 is not the MPU platform that I'm testing on.

> What is stopping us from simply waiting for the `buffer_release` event before making LVGL render the next frame ?

This might be an approach, that's why I've opened the issue, in order to get together to a conclusion for a fix

AndreCostaaa, 2025-07-02:
I have a feeling it's most straightforward approach and as it avoid changing the parts of the core library

kisvegabor, 2025-07-02:
> What is stopping us from simply waiting for the buffer_release event before making LVGL render the next frame ?

Not Wayland expert, just asking: can we do this right after flushing? 

If rendering is fast we might have an idle period between 2 renderings. If it's actually idle in this period and starts doing something right before rendering, we might not fit in a 33ms or 16ms frame.

cosmindanielradu19, 2025-08-29:
When #8563 is merged, this can be closed.