lvgl/lvgl #8540: LV_COLOR_FORMAT_I1 image won't show on monochrome dispay

mutatrum, 2025-07-05:
### LVGL version

v9.3.0

### Platform

ESP32S3 with SSD1306 OLED.

espressif/esp_lvgl_port (2.6.0)
lvgl/lvgl (9.3.0)
idf (5.4.1)

### What happened?

When I define an image with `LV_COLOR_FORMAT_I1`, it won't show on a diplay with color_format LV_COLOR_FORMAT_I1. It's fine if the image is defines as L8 or RGB565, but obviously that's much bigger in rom.

### How to reproduce?

```c
const LV_ATTRIBUTE_MEM_ALIGN LV_ATTRIBUTE_LARGE_CONST LV_ATTRIBUTE_IMAGE_LOGO uint8_t example_image_map[] = {
  0xff, 0xff, 0xff, 0xff,
  0x00, 0x00, 0x00, 0x00,

  0b11111101, 0b11111111, 0b11111110, 0b11111111, 0b11111111, 0b11111111, 0b11111111, 0b11111111, 0b11111111, 0b11111000,
  ... more lines ...
  0b10011111, 0b11111111, 0b11111111, 0b11111111, 0b11111111, 0b11111111, 0b11111111, 0b11111111, 0b11111111, 0b11111000,
};

const lv_image_dsc_t example_image = {
  .header.cf = LV_COLOR_FORMAT_I1,
  .header.magic = LV_IMAGE_HEADER_MAGIC,
  .header.w = 77,
  .header.h = 30,
  .header.stride = 10,
  .data_size = sizeof(bitaxe_logo_map),
  .data = example_image_map,
};
```

Part of the display configuration (SSD1306):
```c
    const lvgl_port_display_cfg_t disp_cfg = {
        .io_handle = io_handle,
        .panel_handle = panel_handle,
        .buffer_size = GLOBAL_STATE->DISPLAY_CONFIG.h_res * GLOBAL_STATE->DISPLAY_CONFIG.v_res,
        .double_buffer = true,
        .hres = GLOBAL_STATE->DISPLAY_CONFIG.h_res,
        .vres = GLOBAL_STATE->DISPLAY_CONFIG.v_res,
        .monochrome = true,
        .color_format = LV_COLOR_FORMAT_I1,
        .flags = {
            .swap_bytes = false,
            .sw_rotate = false,
        }
    };
```

mutatrum, 2025-07-23:
Please keep open. Not sure if i can give more feedback, if so, please let me know.

rovo89, 2025-07-24:
I ran into the same issue and after debugging through the drawing code for one hour, I pinned it down to the fact that `lv_bin_decoder_get_area()` converts the image to `ARGB888`: https://github.com/lvgl/lvgl/blob/639c23cfceeb30bab42388d8e1389cfacd7a09b0/src/libs/bin_decoder/lv_bin_decoder.c#L438-L439

Later, that decoded image is blended back to I1, but that can only be done if `ARGB888` support is enabled:
https://github.com/lvgl/lvgl/blob/639c23cfceeb30bab42388d8e1389cfacd7a09b0/src/draw/sw/blend/lv_draw_sw_blend_to_i1.c#L356-L360

So, long story short, this works for me (had previously only enabled `LV_DRAW_SW_SUPPORT_I1`):

```c++
#define LV_DRAW_SW_SUPPORT_ARGB8888     1
```

No idea whether it could work without that extra step via `ARGB888`... feels like overhead to me, but I don't have any idea of the codebase.

mutatrum, 2025-07-25:
That's a good find, can't test right now, but I'll come back on this in a few weeks. Dont know the internals good enough to determine if it can be done without this step, and if it can, if it's worth it.

At the least it should be added to the docs or it should emit a warning, as this requirement is obscure.

mutatrum, 2025-07-25:
Yes, I disabled all other color formats.

AndreCostaaa, 2025-07-25:
> But I don't think I set up the logs at all, starting to use this on an STM32 with ChibiOS, I don't think there's a builtin way for this. In any case, it will need some external receiver of the logs instead of just popping up in the terminal like it would be for a Linux program. But nothing that LVGL can do about it I suppose.

There's an open PR that adds better ChibiOS support to LVGL so once that is merged it might make it easier to get that up and running

Also this would've happened on the LVGL PC simulator as well, same config would have produced the same problem and would probably be easier to debug. If you don't already use one I highly recommend. Here's the vscode one https://github.com/lvgl/lv_port_pc_vscode/ but there's also ready-to-use repos for eclipse, visual studio or simply linux with cmake

AndreCostaaa, 2025-07-25:
Do you think it would help seeing the exact color format missing? The current warn message says "Unsupported color format"

PR #8620 (2025-08-14): docs(image): document the fact that indexed data is stored as argb8888
Fixes #8540 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-for [... cut]
closed 2025-08-14 by uLipe