lvgl/lvgl #8542: libjpeg-turbo causes an error

jinsc123654, 2025-07-06:
### LVGL version

9.3.0

### Platform

RK3506 BuildRoot(linux)
lvgl9.3.0
libjpeg-turbo main (20250-7-06)

### What happened?

Using libjpeg-turbo on LVGL 9, cloned from the main branch as of 2025-07-06.
If the "Default image cache size. 0 to disable caching" option is disabled, there is no issue. However, when enabling the cache size to 30MB, decoding JPG images will cause the program to crash after the 5th image.

[image]

[image]

[image]

Coloryr, 2025-07-06:
RK的SDK里面的LVGL不是最新的，是9.0.0

jinsc123654, 2025-07-07:
> RK的SDK里面的LVGL不是最新的，是9.0.0

I cloned both LVGL and libjpeg-turbo repositories myself.  This menuconfig interface was set up by myself.

[image]

[image]
The libjpeg library provided by RK was outdated, so I also re-cloned and built a newer version myself.
[image]

jinsc123654, 2025-07-10:
After debugging, I found a bug in the lvgl img cache. It will cause program exceptions when starting free. This not only occurs in libjpeg_turbo, but also in lodepng

[image]

jinsc123654, 2025-07-10:
Bugs caused by [Number of draw units]
[image]

[image]

KyL0N, 2025-07-14:
https://github.com/lvgl/lvgl/issues/8123

same problem?

jinsc123654, 2025-07-17:
😊

AndreCostaaa, 2025-07-17:
Hi, it seems like it's the same problem as #8123

Thank you for opening this issue. The other issue eventually got stale and was automatically closed, i'll mark this as important to avoid the same fate as that one, we need to allocate some time to fix this 

cc @W-Mai

kisvegabor, 2025-07-28:
@W-Mai will you have time to look I to it in the near future?

W-Mai, 2025-07-28:
> @W-Mai will you have time to look I to it in the near future? 

sure, let me take a look at this

AndreCostaaa, 2025-07-29:
Collecting all similar issues to the same place so it's easier to navigate
Same issue here: #8526 

### LVGL version

v9.3.0

### Platform

Eclipse with SDL on Linux.

### What happened?

Image cache process hangs.
There is a problem with lv_cache_class_lru_rb_count, but lv_cache_class_lru_ll_count is normal.

 heap-use-after-free on address 0x6040000e4670 at pc 0x7f8289504aad bp 0x7f8282efb720 sp 0x7f8282efb710
READ of size 8 at 0x6040000e4670 thread T4
    #0 0x7f8289504aac in get_victim_cb /components/lvgl/src/misc/cache/class/lv_cache_lru_rb.c:428
    #1 0x7f82895008ba in cache_evict_one_internal_no_lock /components/lvgl/src/misc/cache/lv_cache.c:329
    #2 0x7f8289500adf in cache_add_internal_no_lock /components/lvgl/src/misc/cache/lv_cache.c:352
    #3 0x7f82894ffa90 in lv_cache_add /components/lvgl/src/misc/cache/lv_cache.c:138
    #4 0x7f82895cb279 in lv_image_decoder_add_to_cache /application_components/lvgl/src/draw/lv_image_decoder.c:225
    #5 0x7f82895f82d5 in decoder_open /application_components/lvgl/src/libs/libpng/lv_libpng.c:184
    #6 0x7f82895cac63 in lv_image_decoder_open /application_components/lvgl/src/draw/lv_image_decoder.c:129
    #7 0x7f8289532644 in lv_draw_image_normal_helper /application_components/lvgl/src/draw/lv_draw_image.c:234
    #8 0x7f828956d733 in lv_draw_sw_image /application_components/lvgl/src/draw/sw/lv_draw_sw_img.c:205
    #9 0x7f82895401b3 in execute_drawing /application_components/lvgl/src/draw/sw/lv_draw_sw.c:404
    #10 0x7f828953fea8 in render_thread_cb /application_components/lvgl/src/draw/sw/lv_draw_sw.c:365
    #11 0x7f828963af63 in generic_callback /application_components/lvgl/src/osal/lv_pthread.c:189
    #12 0x7f828a009608 in start_thread /build/glibc-B3wQXB/glibc-2.31/nptl/pthread_create.c:477
    #13 0x7f8288f2d352 in __clone (/lib/x86_64-linux-gnu/libc.so.6+0x11f352)

0x6040000e4670 is located 32 bytes inside of 40-byte region [0x6040000e4650,0x6040000e4678)
freed by thread T4 here:
    #1 0x7f82895ffe09 in lv_free_core /application_components/lvgl/src/stdlib/clib/lv_mem_core_clib.c:74
    #2 0x7f82895ffb59 in lv_free /application_components/lvgl/src/stdlib/lv_mem.c:136
    #3 0x7f828963cdcb in lv_rb_remove_node /application_components/lvgl/src/misc/lv_rb.c:206
    #4 0x7f82895043af in remove_cb /application_components/lvgl/src/misc/cache/class/lv_cache_lru_rb.c:345
    #5 0x7f828950095a in cache_evict_one_internal_no_lock /application_components/lvgl/src/misc/cache/lv_cache.c:336
    #6 0x7f8289500adf in cache_add_internal_no_lock /application_components/lvgl/src/misc/cache/lv_cache.c:352
    #7 0x7f82894ffa90 in lv_cache_add /application_components/lvgl/src/misc/cache/lv_cache.c:138
    #8 0x7f82895cb279 in lv_image_decoder_add_to_cache /application_components/lvgl/src/draw/lv_image_decoder.c:225
    #9 0x7f82895f82d5 in decoder_open /application_components/lvgl/src/libs/libpng/lv_libpng.c:184
    #10 0x7f82895cac63 in lv_image_decoder_open /application_components/lvgl/src/draw/lv_image_decoder.c:129
    #11 0x7f8289532644 in lv_draw_image_normal_helper /application_components/lvgl/src/draw/lv_draw_image.c:234
    #12 0x7f828956d733 in lv_draw_sw_image /application_components/lvgl/src/draw/sw/lv_draw_sw_img.c:205
    #13 0x7f82895401b3 in execute_drawing /application_components/lvgl/src/draw/sw/lv_draw_sw.c:404
    #14 0x7f828953fea8 in render_thread_cb /application_components/lvgl/src/draw/sw/lv_draw_sw.c:365
    #15 0x7f828963af63 in generic_callback /application_components/lvgl/src/osal/lv_pthread.c:189
    #16 0x7f828a009608 in start_thread /build/glibc-B3wQXB/glibc-2.31/nptl/pthread_create.c:477

previously allocated by thread T4 here:
    #1 0x7f82895ffdc2 in lv_malloc_core /application_components/lvgl/src/stdlib/clib/lv_mem_core_clib.c:64
    #2 0x7f82895ffab2 in lv_malloc_zeroed /application_components/lvgl/src/stdlib/lv_mem.c:100
    #3 0x7f828963d4cd in rb_create_node /application_components/lvgl/src/misc/lv_rb.c:335
    #4 0x7f828963c257 in lv_rb_insert /application_components/lvgl/src/misc/lv_rb.c:76
    #5 0x7f82895031f1 in alloc_new_node /application_components/lvgl/src/misc/cache/class/lv_cache_lru_rb.c:159
    #6 0x7f8289504147 in add_cb /application_components/lvgl/src/misc/cache/class/lv_cache_lru_rb.c:313
    #7 0x7f8289500bc6 in cache_add_internal_no_lock /application_components/lvgl/src/misc/cache/lv_cache.c:355
    #8 0x7f82894ffa90 in lv_cache_add /application_components/lvgl/src/misc/cache/lv_cache.c:138
    #9 0x7f82895cb279 in lv_image_decoder_add_to_cache /application_components/lvgl/src/draw/lv_image_decoder.c:225
    #10 0x7f82895f82d5 in decoder_open /application_components/lvgl/src/libs/libpng/lv_libpng.c:184
    #11 0x7f82895cac63 in lv_image_decoder_open /application_components/lvgl/src/draw/lv_image_decoder.c:129
    #12 0x7f8289532644 in lv_draw_image_normal_helper /application_components/lvgl/src/draw/lv_draw_image.c:234
    #13 0x7f828956d733 in lv_draw_sw_image /application_components/lvgl/src/draw/sw/lv_draw_sw_img.c:205
    #14 0x7f82895401b3 in execute_drawing /application_components/lvgl/src/draw/sw/lv_draw_sw.c:404
    #15 0x7f828953fea8 in render_thread_cb /application_components/lvgl/src/draw/sw/lv_draw_sw.c:365
    #16 0x7f828963af63 in generic_callback /application_components/lvgl/src/osal/lv_pthread.c:189
    #17 0x7f828a009608 in start_thread /build/glibc-B3wQXB/glibc-2.31/nptl/pthread_create.c:477
[... 10 lines cut ...]
  0x0c08800148e0: fa fa 00 00 00 00 00 00 fa fa fd fd fd fd fd fa
[... cut ...]
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==319157==ABORTING

AndreCostaaa, 2025-09-30:
This PR has been merged recently, but I haven't tested to see if it fixes the issue https://github.com/lvgl/lvgl/pull/8735