lvgl/lvgl #8554: Tiny JPEG Decompressor (TJpgDec) fails with ESP32C6

supremeneuron, 2025-07-09:
### LVGL version

v.9.3.0

### Platform

Windows 11 Pro, using Arduino 2.3.6 with ESP32 3.1.3 library for a ESP32C6 Development board connected to an ILI9488 display

### What happened?

Whenever I call the function to load a JPG file, the application crashes with a vague Guru Meditation error.  Now the interesting part is that the same EXACT code compiled for the ESP32C3 works fine?!? 

### How to reproduce?

    //basically the example code provided.

    lv_obj_t * wp;

    wp = lv_image_create(lv_screen_active());
   // Assuming a File system is attached to letter 'A'
    // E.g. set LV_USE_FS_STDIO 'A' in lv_conf.h 
    lv_image_set_src(wp, "A:/test.jpg");
    lv_obj_center(wp);

AndreCostaaa, 2025-07-25:
Hi, i'm sorry I don't follow you

It works on esp32 but not on your windows machine, is that it?

In that case, for windows, should it be?

Note the removal of `/`:
```c
lv_image_set_src(wp, "A:test.jpg");
```

kisvegabor, 2025-09-29:
Maybe I'm too late, but do you see any [Log from LVGL](https://docs.lvgl.io/master/details/debugging/log.html)? Any crash log would be helpful too.

supremeneuron, 2025-09-30:
@kisvegabor , thanks for looking into this issue.  Unfortunately, no I didn't see any logs that appeared meaningful.  Just that seemingly odd Guru Meditation error.

AndreCostaaa, 2025-10-06:
Could be a misconfiguration issue, I'm not very familiar with esp32C3 nor esp32C6. Maybe @liamHowatt, @uLipe or @fbiego could have some ideas.

Can you provide an minimal reproducible example alongside your image and your sdkconfig/lv_conf.h ?

supremeneuron, 2025-10-06:
@AndreCostaaa 

Below is the code in my Arduino program to load the JPG, I have included the BMP code as well which works fine.  

Also I've included the lv_config.h component you asked for where I included the elements that have been altered.

ERROR from the serial monitor:

Rebooting...
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1228
load:0x4086c110,len:0xd9c
load:0x4086e610,len:0x2f74
entry 0x4086c110
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

...after this the device reboots

The code AFTER setting up the display and other elements of the device....

// The following loads JPG  (This does not work)
  lv_obj_t * wp;

    wp = lv_image_create(lv_screen_active());
   // Assuming a File system is attached to letter 'A'
    // E.g. set LV_USE_FS_STDIO 'A' in lv_conf.h 
    lv_image_set_src(wp, "A:/ltest.jpg");
    lv_obj_center(wp);

   

*** if I replace the above code with the code below it works.
   //load a BMP image (THIS WORKS)
   lv_obj_t * img = lv_image_create(lv_screen_active());
    lv_image_set_src(img, "A:360.bmp");
    lv_obj_center(img);

The lv_config.h changes....

#define LV_FS_DEFAULT_DRIVER_LETTER 'A'

/** API for FATFS (needs to be added separately). Uses f_open, f_read, etc. */
#define LV_USE_FS_FATFS 1
#if LV_USE_FS_FATFS
    #define LV_FS_FATFS_LETTER 'A'     /**< Set an upper-case driver-identifier letter for this driver (e.g. 'A'). */
    #define LV_FS_FATFS_PATH ""         /**< Set the working directory. File/directory paths will be appended to it. */
    #define LV_FS_FATFS_CACHE_SIZE 0    /**< >0 to cache this number of bytes in lv_fs_read() */
#endif

/** BMP decoder library */
#define LV_USE_BMP 1

/** JPG + split JPG decoder library.
 *  Split JPG is a custom format optimized for embedded systems. */
#define LV_USE_TJPGD 1