lvgl/lvgl #8555: Circular scrolling label animation will not repeat

apeschock, 2025-07-09:
### LVGL version

v8.4 and up

### Platform

On any platform, even the web examples.

### What happened?

The circular scrolling animation does not repeat at all until you change the text.  lv_anim_set_repeat_delay does not take effect.

lizhaoming634, 2025-07-09:
If the text is wider than the Label, scroll it horizontally continuously. If it's higher, scroll vertically. Only one direction is scrolled and horizontal scrolling has higher precedence. 
Perhaps your text is narrower than the label.

MikeEckels, 2025-07-09:
> If the text is wider than the Label, scroll it horizontally continuously. If it's higher, scroll vertically. Only one direction is scrolled and horizontal scrolling has higher precedence. Perhaps your text is narrower than the label.

I don't believe this is what @apeschock is refereeing to. Scrolling text itself works as it should, and continuously scrolls when `LV_LABEL_LONG_SCROLL_CIRCULAR ` is applied.

This issue is specifically when a custom animation template is applied. The `lv_anim_set_delay()` and `lv_anim_set_repeat_delay()` functions do not work as they should. The animation never repeats after the first scroll.

The example [here](https://github.com/lvgl/lvgl/blob/c1205ad4895f37fef386352bbd7f2b6b54df0afc/examples/widgets/label/lv_example_label_5.c) also shows this incorrect behavior. The web example ["Customize circular scrolling animation"](https://docs.lvgl.io/9.2/widgets/label.html) also does not repeat.

apeschock, 2025-07-25:
That was it, thank you!

PR #8621 (2025-07-29): fix(example/label): make animation repeat indefinitely
Fixes #8555 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`astyle [... cut]