lvgl/lvgl #8558: High CPU Load While Using v9.3 with FreeRTOS

fatihozen3, 2025-07-09:
### Introduce the problem

I use NXP IMXRT1176. I was using v9.2.2, I wanted to try v9.3. I have opened system monitor and performance monitor, I also use FreeRTOS. Then I get below warning, and CPU usage is %100, what may cause this ?

[image]

uLipe, 2025-07-09:
Hello @fatihozen3 I was able to reproduce that, it is an internal issue with the idle percent measurement procedure, so the CPU is not actually consuming near of 100%

fatihozen3, 2025-07-09:
So, may I skip this warning or do I have to do something about it ? And, how can I see actual real CPU load ?

uLipe, 2025-07-09:
For now you can just skip this warning, there will be a patch on the demo project that fixes it.