lvgl/lvgl #8568: Code works fine with 9.2.2 but crash with  9.3.0

Herve14, 2025-07-13:
### LVGL version

9.3.0

### Platform

    MCU: SAM38XE (Arduino DUE)
    Display: RA8875 480x240 RGB565 // mode
    Compiler: ARM GCC via PlatformIO
    External Memory: no
    Graphics Layer: no
    LVGL version: 9.3
    OS: no

### What happened?

The code works fine with 9.2.2 but crash with 9.3.0 with tab widget.

I get the following message :

[Error] (43.460, +43460)         lv_tabview_get_tab_active: Asserted at expression: obj != NULL (NULL pointer) lv_tabview.c:240

AndreCostaaa, 2025-07-14:
Hi @Herve14

Do you have a stack trace you can provide? A snippet of code reproducing the error would be great too

kisvegabor, 2025-07-28:
@AndreCostaaa can you take a look at it?

Herve14, 2025-08-26:
Hi André
I'm sorry, but I think there is still a problem with the TabView widget. I'll open a new post to describe it.
Best regards