lvgl/lvgl #8603: CI Failure on Hardware Build Test

vwheeler63, 2025-07-23:
### LVGL version

v9.3

### Platform

GitHub

### What happened?

Apparently all current PRs being submitted have CIs failing with this error:

[image]

### How to reproduce?

Submit any PR based on current `master` branch.

AndreCostaaa, 2025-07-24:
I will look into cancelling the workflow if there are no builders available instead of exiting with an error :)

vwheeler63, 2025-07-24:
> I will look into cancelling the workflow if there are no builders available instead of exiting with an error :)

Okay.  How come OTHER CI tests don't fail for that reason?

AndreCostaaa, 2025-07-24:
Oh, I understand where the confusion is coming from

That CI test is the one that builds LVGL for the hardware farm we deployed last week. In that context, we're talking about an EJ Builder - https://embj-org.github.io/ej/ 

LVGL has one EJ Builder deployed but is currently offline because it's being used for development purposes by @liamHowatt 

The other CI tests are run in Github's hardware which is available

vwheeler63, 2025-07-25:
> Oh, I understand where the confusion is coming from

That's pretty nifty!  So the hardware would have to sit plugged into its "programmer" with something running on the PC to intercept programming/test requests.  Does it re-flash each time a test is run, or is the hardware pre-programmed with a test program and reports the results?

AndreCostaaa, 2025-07-25:
It does! The benchmark application is always the same but the underlying LVGL version is updated so that we can do performance comparisons between different commits

It also supports multiple configurations per board so the same board could be flashed multiple times for each tested version

vwheeler63, 2025-07-25:
> It does! The benchmark application is always the same but the underlying LVGL version is updated so that we can do performance comparisons between different commits
> 
> It also supports multiple configurations per board so the same board could be flashed multiple times for each tested version

That's an impressive feature!  I look forward to learning about it when I study GitHub's CI stuff.

AndreCostaaa, 2025-07-25:
It's completely unrelated to Github CI actually, it's all EJ in this case

Even thought similar stuff could be done using Github Actions, EJ just does it in an independent that doesn't even need to be connected to the CI

In this case we simply use github to tell ej "time to run a test for this version". Does it make sense?

vwheeler63, 2025-07-26:
> Does it make sense?

Yes!  So what I am understanding (I think) is that GitHub Actions simply launches EJ which takes care of the connection, interaction, and reporting (success, failure, comment).

closed 2025-07-26 by vwheeler63