lvgl/lvgl #8650: LVGL 8.x vs 9.x broke my brain and my project

SpiritOwl, 2025-08-01:
### Introduce the problem

Hi team,

I’ve spent hours today attempting to get a basic example working with a LilyGO T-Display-S3 using LVGL. I followed various instructions, downgraded to v8.x, upgraded again to v9.x, and kept hitting circular errors — broken compatibility, renamed types like `lv_disp_draw_buf_t` vs `lv_draw_buf_t`, missing `lv_conf.h`, etc.

The Arduino support seems caught in a confusing middle state between v8 and v9, and it's extremely difficult to know what version *anything* is supposed to work with.

At the end of it, I feel incredibly frustrated and disappointed with how difficult it was to get something basic working. I’ve been in the embedded/dev world for years — this wasn’t a novice mistake. The documentation and example ecosystem clearly hasn’t caught up.

Please take this as constructive feedback: something needs to be done to help developers avoid this version mismatch hell. If there’s an officially recommended version for Arduino and ESP32, make it obvious — and lock everything to it (examples, tutorials, libraries, etc.). And I'm not an expert, just a tinkerer. 

Sincerely,  
A disappointed (but once excited) dev.

cc: @kisvegabor

SpiritOwl, 2025-08-01:
[examples/factory](https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/examples/factory)

vwheeler63, 2025-08-01:
Hi, @SpiritOwl !

Were you aware of this v8->v9 migration guide?   https://docs.lvgl.io/9.0/CHANGELOG.html#migration-guide

vwheeler63, 2025-08-22:
> [@vwheeler63](https://github.com/vwheeler63) , the lv_api_map link is broken. Seems there are now several - does the version number mean the upgrade-from or upgrade-to? Is is expected that within minor version of LVGL v9 there are type changes?

Hi, @bryghtlabs-richard !

1. I think you will find that in various versions of v9, these are included automatically.
2. Each one means "upgrade to".
3. From v9.0 to v9.3, there are a few type changes (fields within structs, mostly should not be being accessed directly by user code), but mostly name changes.
4. v8 to v9, there are a lot of changes.  I recommend referring to [this migration guide](https://docs.lvgl.io/9.0/CHANGELOG.html#migration-guide).  Note that from the normal docs website, you have to select v9.0 docs and then this is found within the CHANGELOG for that version.

AndreCostaaa, 2025-08-25:
It's expected to have breaking changes between two major versions, these changes are necessary and what allow libraries to make big changes that allow for better features in the future. The contract with the developers is to make sure we mark these big incompatible versions with an actual MAJOR number change (v8 vs v9)

We always recommend using the latest version which is currently `v9.3.0`: https://docs.arduino.cc/libraries/lvgl/. We also have some integration specific documentation for it: https://docs.lvgl.io/master/integration/framework/arduino.html`

Do you have some specific issues for something that went wrong when trying to use a specific version? If so we're happy to help, but note that you shouldn't ever expect code that worked with v8 to work with v9

bryghtlabs-richard, 2025-08-25:
> It's expected to have breaking changes between two major versions, these changes are necessary and what allow libraries to make big changes that allow for better features in the future.

Agreed. However, it seems some fraction of the changes I'm integrating seem to be naming-churn rather than functional changes. This makes it more difficult to maintain code that supports projects on multiple LVGL versions.

> Do you have some specific issues for something that went wrong when trying to use a specific version?

None of the individual issues are difficult to integrate. Most of our LVGL code is auto-generated by a desktop tool that is version-aware, but so far I've updated 160/220 handwritten files.

> you shouldn't ever expect code that worked with v8 to work with v9

I'd like to suggest that API stability be considered as a feature eventually. Not as a hard requirement, just that when code could work between versions, it would be nice if it did without porting. At within least minor versions.

AndreCostaaa, 2025-08-25:
> I'd like to suggest that API stability be considered as a feature eventually. Not as a hard requirement, just that when code could work between versions, it would be nice if it did without porting. At within least minor versions.

That's already the case, you shouldn't face breaking changes across minor version changes, code that works for v9.0 should work for v9.3

closed 2025-09-24 by lvgl-bot