lvgl/lvgl #8651: The SVG decoder has a serious memory leak.

iamshiqing, 2025-08-01:
### LVGL version

v9.3.0

### Platform

VSCode with SDL on windows 

### What happened?

Display an SVG image using lv_image

### How to reproduce?

lv_obj_t *widget = lv_image_create(lv_screen_active());
lv_image_set_src(widget, "S:tiger.svg");

iamshiqing, 2025-08-01:
After the program starts running, continuously moving the mouse over the SVG image causes a memory leak.By reading lv_svg_decoder.c, I found that the memory leak was caused because the svg_draw callback function did not release draw_buf after rendering. I added
(draw_buf->handlers->buf_free_cb)((void *)list);
at the end of the svg_draw function, and after trying again, no memory leak occurred.

vwheeler63, 2025-08-01:
@iamshiqing Wow, this is important!  Would you mind submitting a PR targeted for the `master` branch for review (assuming you are working with the latest code)?

closed 2025-08-25 by lvgl-bot

kisvegabor, 2025-09-27:
Fixed by https://github.com/lvgl/lvgl/pull/8658