lvgl/lvgl #8664: EVE images on GPU Flash

george-palmsens, 2025-08-05:
### Introduce the problem

Apologies for a second issue, I shift+enter'd and github submitted and closed my original one... 🤦 

I am planning to contribute on top of the excellent work of @liamHowatt in #8231.

That PR adds support for the EVE series of embedded "GPUs" - SPI connected devices that accept a queue of draw commands instead of a pixel buffer.

It rules, but it's missing a feature which I personally critically need - Utilising the GPU's FLASH storage.

The EVE chips can be connected directly to a flash chip, where they can have graphical assets stored. Then to draw them on screen, you merely have to send a reference to their address on the flash.

Without this, the EVE chip is incapable of displaying large images, since their raw bitmap representation does not fit in its internal RAM.

I have made a proof of concept locally for this, but I would like some advice on whether that approach is suitable to productionise and contribute.

### Proposal

- Provide a function `lv_draw_eve_flash_storage_register` which sets up this system. 
  - Register a filesystem to represent the flash
    - This filesystem would not succeed on any actual read or write operations
  - Register an image decoder that accepts any image on that filesystem, and marks it as requiring custom draw commands
- In those custom draw commands, write the necessary EVE commands to draw a flash image.

So something like:

```c
static lv_draw_eve_flash_storage_t flash;
lv_draw_eve_flash_storage_register(&flash, 'Z');
lv_draw_eve_flash_storage_add_image(&flash, "img_1", IMG_1_ADDR, /* other relevant image params */);

/*
...
*/

lv_image_set_src(img1, "Z:/img_1");
```

Limitations:

Other things can be on the external flash, like animations. I don't have the experience with them to know how they would be best included in this proposal.

---

I feel like this approach is quite forced, and doesn't feel "right". I would rather I could attach some driver specific metadata to an image object that would get transparently passed through. IE:

```c
lv_obj_t *img1 = lv_image_create(lv_screen_active());
lv_draw_eve_flash_image_data_t const data = {
  .addr = 0xdeadbeef,
  .encoding = EVE_L8,
  .width = 100,
  .height = 50,
  .stride = 200
};
lv_draw_eve_set_flash_image_data(img1, &data);
```

george-palmsens, 2025-08-05:
A better option, I think, is not to use a custom draw command but rather to permit the EVE draw unit to understand when an image that originates from a file is registered with the EVE flash pseudo-filesystem, and render it accordingly.

liamHowatt, 2025-08-28:
Ah I see. There's only one copy in flash and LVGL takes your parameters about the images on faith. Your `lv_draw_eve_flash_image_data_t` makes sense now. I think it could be an `lv_image_header_t` and an explicit address, as you say. This mode of operation is distinctly different from what I'd said about LVGL uploading its copy of some assets.

I think your idea of a virtual file path approach was good foresight. I think draw_eve would need to register both a filesystem driver _and_ an image decoder.

1. The new registered FS driver would cause a wrapper struct to be read out with a unique header magic.
2. The new registered image decoder would look for the unique header magic of the wrapper struct and unwrap the inner `lv_image_header_t` and claim image decoded successfully and provide a bogus `data` pointer.
3. When the image header gets to `draw_eve`, it will do some check on the data pointer to determine if it's a special bogus one, which contains the EVE flash address somehow. And then draw_eve is now free to command EVE to render the image from flash.

I'm open to a simpler way. It's just what I could think of 😅.

Would you use a build-time tool to make the assets ARGB4444 for the EVE flash?

george-palmsens, 2025-08-29:
Aha sounds a bit scary! But I've seen the "magic header" approach used in the image code so I guess it's consistent.

We have an entire build-time pipeline for asset creation - In general, I don't think LVGL can be expected to generate what the user wants in flash. For example I might want to add audio, other encodings and so forth.

So I'm fine with leaving all that up to the user, not part of LVGL.

capricorn-one, 2025-08-30:
Hey just wanted to throw my hat in here if you want some help with this.

I actually implemented this a little over a year ago on a fork someone made of pre-release lvgl 9. See what I did here:

https://github.com/juanjqh/lvgl_eve_gpu_test-main/issues/4

Obviously, I had to butcher LVGL to make this work, but would love to do it the right way if you have some ideas and want some help. I'm familiar enough with the EVE side to know what to do, and don't want to have to rely on an LVGL fork in the future. LMK if you've made any progress on this or if it was just a feature request.

Looks like I took a similar approach to what you're suggesting, basically making a custom descriptor for flash images. It did work, but it's ugly. Happy to contribute to a better solution, but can confirm, when it works, it works great!

It's been a while now since I've looked at the code, but if you're curious how I did this, the added eve_img_flash drawing functions can be found here:

https://github.com/capricorn-one/lvgl/blob/master/src/draw/eve/lv_draw_eve_image.c