lvgl/lvgl #8691: SVG rendering fails with LV_DISPLAY_RENDER_MODE_PARTIAL in LVGL v9.3

lijingjie, 2025-08-14:
### LVGL version

v9.3.0

### Platform

WSL
Linux test-PC 5.15.167.4-microsoft-standard-WSL2 #1 SMP Tue Nov 5 00:21:55 UTC 2024 x86_64 x86_64 x86_64 GNU/Linux

### What happened?

When using the SDL backend width LV_DISPLAY_RENDER_MODE_PARTIAL,SVG images fail to render on screen. 
Switching to LV_DISPLAY_RENDER_MODE_DIRECT resolves the issue.

The different between the LV_DISPLAY_RENDER_MODE_DIRECT and LV_DISPLAY_RENDER_MODE_PARTIAL is the y_offset value as the code belows.
In LV_DISPLAY_RENDER_MODE_DIRECT mode, the y_offset value is always 0, but In LV_DISPLAY_RENDER_MODE_PARTIAL mode, the y_offset value is ​​non-zero​​.

Manually forcing y_offset = 0 in PARTIAL mode successfully renders SVG images.

lv_draw_sw_vector.c
```
static void _task_draw_cb(void * ctx, const lv_vector_path_t * path, const lv_vector_draw_dsc_t * dsc)
{
    _tvg_draw_state * state = (_tvg_draw_state *)ctx;
    Tvg_Canvas * canvas = (Tvg_Canvas *)state->canvas;

    Tvg_Paint * obj = tvg_shape_new();

    int32_t y_offset = state->partial_y_offset;

    _tvg_rect rc;
    lv_area_to_tvg(&rc, &dsc->scissor_area);
```

### How to reproduce?

lv_obj_t * img14 = lv_image_create(lv_layer_top());
lv_obj_set_size(img14, (300), (300));
lv_obj_set_pos(img14, (0), (0));
lv_image_set_src(img14, "A:xxxx.svg");

lijingjie, 2025-09-01:
@kisvegabor

kisvegabor, 2025-09-27:
@liamHowatt please take a look at this.

liamHowatt, 2025-09-28:
Hey, I couldn't preproduce it. I tried `master` and `release/v9.3`.

Is there anything else I'm missing?

SVG: https://github.com/user-attachments/assets/5b3c373e-891f-4581-a8c5-bc37045b31f0

I used:

```c
    lv_obj_t * svg = lv_image_create(lv_screen_active());
    lv_obj_set_size(svg, (300), (300));
    lv_obj_set_pos(svg, (0), (0));
    lv_image_set_src(svg, "A:compass-ruler.svg");
```

```
LV_USE_SDL 1
LV_SDL_RENDER_MODE      LV_DISPLAY_RENDER_MODE_PARTIAL
LV_SDL_BUF_COUNT  1   # also tried 2

LV_USE_THORVG_INTERNAL 1
LV_USE_VECTOR_GRAPHIC 1
LV_USE_MATRIX 1
LV_USE_FLOAT 1
LV_USE_SVG 1
LV_USE_VG_LITE_THORVG 1
LV_DRAW_BUF_ALIGN 64
LV_USE_DRAW_VG_LITE 1
LV_VG_LITE_USE_GPU_INIT 1
LV_USE_DEMO_VECTOR_GRAPHIC 1
```

lijingjie, 2025-09-30:
@liamHowatt 
I think the big difference is that, i used the LV_USE_DRAW_SW 1, not LV_USE_DRAW_VG_LITE 1 and LV_USE_VG_LITE_THORVG 1.

Can you try it again use the configurations of below.
LV_USE_VG_LITE_THORVG 0
LV_USE_DRAW_VG_LITE 0
LV_VG_LITE_USE_GPU_INIT 0
LV_USE_DRAW_SW 1