lvgl/lvgl #8697: Changing translation language at runtime

larsk2009, 2025-08-15:
### LVGL version

v9.4.0 (c4887d8adfa75143e923f367694aaa718bca9384)

### Platform

WSL with wayland

### What happened?

Not really a bug, but more of a question of behaviour. I am trying out the new translation methods. I use `lv_translation_set_language` to set the language and `lv_tr` to get a translated string. This works fine. Now when the user wants to change the language in runtime, it seems I need to call `lv_label_set_text` (or equivalent) again with `lv_tr`. I just want to confirm that this is intented and that I am not missing some other option to update the translation in all widgets without having to call `lv_label_set_text` troughout the whole application.

AndreCostaaa, 2025-08-31:
Hi, if my memory serves me right you can simply invalidate the whole screen so that the changes take effect, the labels will get rerendered again with their new value. Let me know if that solves it for you

xlbao123, 2025-09-04:
As I know, there are two possible approaches here:
1. Application-level wrapping:
Create a small wrapper around lv_label_set_text() (or similar APIs). In that wrapper, keep track of all objects and their associated translation keys. When the language changes, you can loop over this list and call lv_label_set_text(obj, lv_tr(key)) again. 
2. Core modification:
Extend lv_label (and possibly other text-bearing widgets) with an additional field like const char *translation_key. Add a new API such as lv_label_set_text_key. At render time, the label would call lv_tr to fetch the current language string. Switching the language would then only require lv_translation_set_language() plus an lv_obj_invalidate() to trigger redraws. This is more convenient but requires patching LVGL.

XuNeo, 2025-09-04:
We could introduce a core event for translation. Any widget that needs translation would handle this event.

The event would carry the target language, and widgets could then call a `gettext`-style API as needed.

With this approach, `lv_translation_set_language` would only need to send the event (with the target language) to the root object.

larsk2009, 2025-10-23:
@AndreCostaaa I saw this PR (https://github.com/lvgl/lvgl/pull/8948) in the changelog of 9.4. Does that mean we can now actually bind translations to labels and we don't have to use the implementation of #8882?