lvgl/lvgl #8719: Circle not rendered properly at top left edge

Finomnis, 2025-08-20:
### LVGL version

v9.3.0, rev b03edc8e6 (Zephyr 4.2.0 release)

### Platform

native_sim/native/64 on Zephyr

### What happened?

When rendering a circle, the four corners are missing a spot.

### How to reproduce?

- Checkout zephyr, according to https://docs.zephyrproject.org/latest/develop/getting_started/index.html
- Create `samples/modules/lvgl/screen_transparency/boards/native_sim_native_64.overlay` and fill it with:
  ```
  &sdl_dc {
      width = <640>;
      height = <480>;
  };
  ```
- Edit `samples/modules/lvgl/screen_transparency/prj.conf` and modify:
  ```
  CONFIG_MAIN_STACK_SIZE=8192
  CONFIG_LV_Z_MEM_POOL_SIZE=65536
  ```
- Edit `samples/modules/lvgl/screen_transparency/src/main.c` and replace the content of `initialize_gui()`:
  ```
  static void initialize_gui(void)
  {
  	lv_coord_t h = lv_obj_get_height(lv_screen_active());
  	lv_obj_set_style_bg_color(lv_screen_active(), lv_color_black(), 0);
  
  	lv_obj_t *circle = lv_obj_create(lv_screen_active());
  	lv_obj_set_style_pad_all(circle, 0, 0);
  	lv_obj_set_style_radius(circle, LV_RADIUS_CIRCLE, 0);
  	lv_obj_set_style_bg_opa(circle, 0, 0);
  	lv_obj_set_style_border_color(circle, lv_color_white(), 0);
  	lv_obj_set_style_border_width(circle, 1, 0);
  	lv_obj_align(circle, LV_ALIGN_CENTER, 0, 0);
  	lv_obj_set_size(circle, h, h);
  }
  ```
- Run `west build -b native_sim/native/64 samples/modules/lvgl/screen_transparency/ -t run -p auto`

Get the following result:

[image]

Zoomed in:

[image]

Finomnis, 2025-09-04:
I don't think anything changed yet.

kisvegabor, 2025-09-29:
Size and border with.

Finomnis, 2025-11-29:
Not stale. And sorry @kisvegabor, somehow missed your response.

Finomnis, 2025-12-10:
@kisvegabor Fair :)