lvgl/lvgl #8802: lv_st7796_create

Wacboss1, 2025-09-03:
### LVGL version

v9.4.0

### Platform

Visual Studio Code with WSL (ubuntu) and cross compiling using Cmake and Clang to an Raspberry Pi Zero 2 W. 

My goal is to use a SPI display with an ST7796S chip to display the lvgl demo.

### What happened?

I'm trying to use the st7796 driver but keep getting the following error when compiling 

```
/home/wclemms/lvgl_projects/beagle_port/main.c:39:9: error: incompatible function pointer types passing 'int32_t (lv_display_t *, const uint8_t *, size_t, const uint8_t *, size_t)' (aka 'int (struct _lv_display_t *, const unsigned char *, unsigned long, const unsigned char *, unsigned long)') to parameter of type 'lv_st7796_send_cmd_cb_t' (aka 'void (*)(struct _lv_display_t *, const unsigned char *, unsigned long, const unsigned char *, unsigned long)') [-Wincompatible-function-pointer-types]
   39 |         my_lcd_send_cmd,
      |         ^~~~~~~~~~~~~~~
/home/wclemms/lvgl_projects/beagle_port/lvgl/src/drivers/display/st7796/lv_st7796.h:48:57: note: passing argument to parameter 'send_cmd_cb' here
   48 |                                 lv_st7796_send_cmd_cb_t send_cmd_cb, lv_st7796_send_color_cb_t send_color_cb);
      |                                                         ^
/home/wclemms/lvgl_projects/beagle_port/main.c:40:9: error: incompatible function pointer types passing 'int32_t (lv_display_t *, const uint8_t *, size_t, uint8_t *, size_t)' (aka 'int (struct _lv_display_t *, const unsigned char *, unsigned long, unsigned char *, unsigned long)') to parameter of type 'lv_st7796_send_color_cb_t' (aka 'void (*)(struct _lv_display_t *, const unsigned char *, unsigned long, unsigned char *, unsigned long)') [-Wincompatible-function-pointer-types]
   40 |         my_lcd_send_color);
      |         ^~~~~~~~~~~~~~~~~
/home/wclemms/lvgl_projects/beagle_port/lvgl/src/drivers/display/st7796/lv_st7796.h:48:96: note: passing argument to parameter 'send_color_cb' here
   48 |                                 lv_st7796_send_cmd_cb_t send_cmd_cb, lv_st7796_send_color_cb_t send_color_cb)
```

My main.c currently looks like this:

```

#include "lvgl/lvgl.h"
#include "lvgl/demos/lv_demos.h"
#include <unistd.h>
#include "build/config.h" //just here to prevent IDEs from getting angry
#include "main.h"
#include "lib/st7796.h"
#include "lib/test_fill.h"

lv_lcd_send_cmd_cb_t my_lcd_send_cmd(
    lv_display_t *disp, 
    const uint8_t *cmd, 
    size_t cmd_size, 
    const uint8_t *param,
     size_t param_size){
        send_cmd(*cmd);
        send_data(param, param_size);
}

lv_lcd_send_color_cb_t my_lcd_send_color(
    lv_display_t *disp, 
    const uint8_t *cmd,
    size_t cmd_size,
    uint8_t *param,
    size_t param_size){
        send_cmd(*cmd);
        send_data(param, param_size);
        return 0;
}

int main(){
    //TODO: write my own drivers because the provided ones don't work
    gpio_setup();
    lv_init();
    lv_display_t* disp = lv_st7796_create(
        HORZ_RES, 
        VERT_RES, 
        LV_LCD_FLAG_NONE, 
        my_lcd_send_cmd,
        my_lcd_send_color);

    lv_demo_widgets();
    lv_demo_widgets_start_slideshow();

    // lv_demo_benchmark();

    uint32_t idle_time;
    /* Handle LVGL tasks */
    while (true) {
        /* Returns the time to the next timer execution */
        idle_time = lv_timer_handler();
        usleep(idle_time * 1000);
    }
}
```

I'm aware there are probably better cleaner ways of doing this but at the moment I am just trying to get it to compile.

AndreCostaaa, 2025-09-03:
Hi, these functions:

```c
lv_lcd_send_cmd_cb_t my_lcd_send_cmd(
    lv_display_t *disp, 
    const uint8_t *cmd, 
    size_t cmd_size, 
    const uint8_t *param,
     size_t param_size){
        send_cmd(*cmd);
        send_data(param, param_size);
}

lv_lcd_send_color_cb_t my_lcd_send_color(
    lv_display_t *disp, 
    const uint8_t *cmd,
    size_t cmd_size,
    uint8_t *param,
    size_t param_size){
        send_cmd(*cmd);
        send_data(param, param_size);
        return 0;
}
```

Should have void as their return type:

```c
void my_lcd_send_cmd(
    lv_display_t *disp, 
    const uint8_t *cmd, 
    size_t cmd_size, 
    const uint8_t *param,
     size_t param_size){
        send_cmd(*cmd);
        send_data(param, param_size);
}

void my_lcd_send_color(
    lv_display_t *disp, 
    const uint8_t *cmd,
    size_t cmd_size,
    uint8_t *param,
    size_t param_size){
        send_cmd(*cmd);
        send_data(param, param_size);
        return 0;
}
```

As per the definition in `lv_lcd_generic_mipi.h`:

```c
typedef void (*lv_lcd_send_cmd_cb_t)(lv_display_t * disp, const uint8_t * cmd, size_t cmd_size, const uint8_t * param,
                                     size_t param_size);
typedef void (*lv_lcd_send_color_cb_t)(lv_display_t * disp, const uint8_t * cmd, size_t cmd_size, uint8_t * param,
                                       size_t param_size);
```

closed 2025-09-03 by AndreCostaaa

Wacboss1, 2025-09-03:
Seems like that worked. Thank you!

AndreCostaaa, 2025-09-03:
No worries. Happy coding 🚀

Wacboss1, 2025-09-03:
Does the documentation need to be updated?

On the website is say that the custom functions should be int32_t instead of void.
https://docs.lvgl.io/master/details/integration/external_display_controllers/st7796.html