lvgl/lvgl #8812: lv_lcd_generic_mipi requires tick before main loop

Wacboss1, 2025-09-04:
### LVGL version

9.4.0

### Platform

I am using the ST7796 drivers to run a SPI display from a Raspberry Pi Zero 2 W. 

### What happened?

After doing a bunch of debugging to figure out why my code was freezing and never reaching the main loop. I discovered that ```init()``` of lv_lcd_generic_mipi.c makes a call to ```lv_delay_ms()```. Therefore i cannot create my display until ticking has started. 

```
static void init(lv_lcd_generic_mipi_driver_t * drv, lv_lcd_flag_t flags)
{
    drv->x_gap = 0;
    drv->y_gap = 0;

    /* init color mode and RGB order */
    drv->madctl_reg = flags & LV_LCD_FLAG_BGR ? LV_LCD_BIT_RGB_ORDER__BGR : LV_LCD_BIT_RGB_ORDER__RGB;
    drv->colmod_reg = flags & LV_LCD_FLAG_RGB666 ? LV_LCD_PIXEL_FORMAT_RGB666 : LV_LCD_PIXEL_FORMAT_RGB565;

    /* init orientation */
    drv->mirror_x = flags & LV_LCD_FLAG_MIRROR_X;
    drv->mirror_y = flags & LV_LCD_FLAG_MIRROR_Y;
    drv->swap_xy = false;
    /* update madctl_reg */
    set_swap_xy(drv, drv->swap_xy);
    set_mirror(drv, drv->mirror_x, drv->mirror_y);

    /* enter sleep mode first */
    send_cmd(drv, LV_LCD_CMD_ENTER_SLEEP_MODE, NULL, 0);
    lv_delay_ms(10);

    /* perform software reset */
    send_cmd(drv, LV_LCD_CMD_SOFT_RESET, NULL, 0);
    lv_delay_ms(200);

    /* LCD goes into sleep mode and display will be turned off after power on reset, exit sleep mode first */
    send_cmd(drv, LV_LCD_CMD_EXIT_SLEEP_MODE, NULL, 0);
    lv_delay_ms(300);

    send_cmd(drv, LV_LCD_CMD_ENTER_NORMAL_MODE, NULL, 0);

    send_cmd(drv, LV_LCD_CMD_SET_ADDRESS_MODE, (uint8_t[]) {
        drv->madctl_reg,
    }, 1);
    send_cmd(drv, LV_LCD_CMD_SET_PIXEL_FORMAT, (uint8_t[]) {
        drv->colmod_reg,
    }, 1);
    send_cmd(drv, LV_LCD_CMD_SET_DISPLAY_ON, NULL, 0);
}
```

From what I understand from the documentation, ticking is suppose to be handled by the lv_timer_handler function in the main loop. So, I'm struggling to see how I can have a tick going every second before reaching the main loop.

I'm currently trying to overcome this by creating a POSIX timer that periodically calls ```lv_tick_inc()```
Here is my current code.
```
#define _POSIX_C_SOURCE 199309L
#include "lvgl/lvgl.h"
#include "lvgl/demos/lv_demos.h"
#include <unistd.h>
#include <signal.h>
#include <sys/time.h>
#include <string.h>
#include "build/config.h" //just here to prevent IDEs from getting angry
#include "main.h"
#include "lib/test_fill.h"

// testing libraries
#include <stdio.h>
#include <stdlib.h>

/* Signal handler for the timer */
void tick_handler(int signum)
{
    (void)signum; /* Unused */
    lv_tick_inc(5); /* Increment LVGL tick by 1ms */
}

// AI generated: 
void setup_tick_timer(void)
{
    struct sigaction sa;
    struct itimerval timer;

    /* Install tick_handler as the signal handler for SIGALRM. */
    memset(&sa, 0, sizeof(sa));
    sa.sa_handler = &tick_handler;
    sigaction(SIGALRM, &sa, NULL);

    /* Configure the timer to expire after 5ms... */
    timer.it_value.tv_sec = 0;
    timer.it_value.tv_usec = 5000; /* 5ms */

    /* ... and every 5ms after that. */
    timer.it_interval.tv_sec = 0;
    timer.it_interval.tv_usec = 5000; /* 5ms */

    /* Start the timer. */
    setitimer(ITIMER_REAL, &timer, NULL);
}

void my_lcd_send_cmd(
    lv_display_t *disp, 
    const uint8_t *cmd, 
    size_t cmd_size, 
    const uint8_t *param,
    size_t param_size)
{
    printf("In send cmd\n");
    (void) disp; /*Unused*/
    for(size_t i = 0; i < cmd_size; i++) {
        send_cmd(cmd[i]);
    }

    printf("Sent cmds, sending data if any\n");
    if(param_size > 0) {
        send_data(param, param_size);
    }
    printf("Sent data if any\n");
}

void my_lcd_send_color(
    lv_display_t *disp, 
    const uint8_t *cmd,
    size_t cmd_size,
    uint8_t *param,
    size_t param_size)
{
    (void) disp; /*Unused*/
    for(size_t i = 0; i < cmd_size; i++) {
        send_cmd(cmd[i]);
    }
    if(param_size > 0) {
        send_data(param, param_size);
    }
}

int main()
{
    printf("Starting SPI\n");
    spi_init(SPI_DEV, SPI_SPEED);
    printf("Setting up GPIO\n");
    gpio_setup();
    printf("Initializing display\n");
    setup_tick_timer();
    lv_init();
    printf("Running minimal ST7796 init\n");
    lv_display_t* disp = lv_st7796_create(
        HORZ_RES, 
        VERT_RES, 
        LV_LCD_FLAG_NONE, 
        my_lcd_send_cmd,
        my_lcd_send_color);

    printf("Starting display init\n");
   // lv_display_t* disp = lv_display_create(HORZ_RES, VERT_RES);

    // lv_display_set_rotation(disp, LV_DISPLAY_ROTATION_90);
    // lv_display_t * disp = lv_linux_fbdev_create();
    // lv_linux_fbdev_set_file(disp, "/dev/fb0");
    
    // lv_obj_t * label;
    // lv_obj_t * button = lv_button_create(lv_screen_active());
    // lv_obj_center(button);
    // label = lv_label_create(button);
    // lv_label_set_text(label, BUTTON_TEXT);
    // lv_obj_center(label);

   
    // lv_demo_widgets();
    // lv_demo_widgets_start_slideshow();

    // lv_demo_benchmark();

    //uint32_t idle_time;
    /* Handle LVGL tasks */
    printf("Entering main loop\n");
    while (true) {
        /* Returns the time to the next timer execution */
        // idle_time = lv_timer_handler();
        // sleep(idle_time * 1000);
    }
}
```

The POSIX timer at least prevents me from hanging but

Wacboss1, 2025-09-05:
So after doing so more research, I learned that ```lv_tick_inc()``` is responsible for keeping track of time for LVGL and should be used outside of the main loop.

```lv_timer_handler()``` is responsible for dealing with events and devices based on how much time has passed. So now I think that my solution of using a POSIX timer is fine.

AndreCostaaa, 2025-09-05:
Hi @Wacboss1 

For POSIX, we recommend setting up a tick callback like this:

```c
static uint32_t tick_get_cb(void)
{
    struct timespec t;
    clock_gettime(CLOCK_MONOTONIC, &t);
    uint64_t time_ms = t.tv_sec * 1000 + (t.tv_nsec / 1000000);
    return time_ms;
}
```

And telling LVGL to use it with:

```c
    lv_tick_set_cb(tick_get_cb);
```
After that, LVGL will always use that function to get the current time and you don't have to call `lv_tick_inc` nor have another thread running in parallel

I hope this helps, let me know if you have any other questions

closed 2025-09-05 by AndreCostaaa