lvgl/lvgl #8813: ESP32-S3 Arduino v3.3.0 with the macro LV_OS_FREERTOS enabled in 100% CPU usage.

LusterWong, 2025-09-05:
### LVGL version

v9.3

### Platform

VSCODE，PlatformIO，ESP32-S3，Arduino V3.3.0
LCD: ST7789 320 x 240

### What happened?

I wanted to run benchmark tests, so I used the [lv_platformio](https://github.com/lvgl/lv_platformio) project. After making modifications, I successfully adapted it for ESP32-S3 and ST7789, with good display and CPU usage. Then, I tried setting LV_USE_OS to LV_OS_FREERTOS. After running, The test results remain the same as before, but the CPU usage consistently shows 100%. I tried to create a task and run lv_timer_handler() inside it, but the result remained the same—100% CPU usage.

### How to reproduce?

This is platformIO.ini file
```
[env]
build_flags =
  ; Don't use lv_conf.h. Tweak params via platfom.ini.
  -D LV_CONF_SKIP
  -D LV_CONF_INCLUDE_SIMPLE
  ; Enable LVGL demo, remove when working on your own project
  -D LV_USE_OS=LV_OS_FREERTOS
  -D LV_USE_FREERTOS_TASK_NOTIFY=1
  -D LV_USE_DEMO_WIDGETS=1
  -D LV_USE_DEMO_BENCHMARK=1
  -D LV_DEMO_BENCHMARK_ALIGNED_FONTS=1
  -D LV_DEMO_BENCHMARK_COLOR_DEPTH=16
  -D LV_DEF_REFR_PERIOD=25
  -D LV_COLOR_DEPTH=16
  -D LV_USE_SYSMON=1
  -D LV_USE_PERF_MONITOR=1

  ; Add more defines below to overide lvgl:/src/lv_conf_simple.h
lib_deps =
  ; Use direct URL, because package registry is unstable
  lvgl@9.3
lib_archive = false
```

I tried used task to run lv timer but no good.. CPU 100% forever
```
void lvgl_task(void *pvParameter)
 {
    while (1)
    {
        lv_timer_handler();
        vTaskDelay(5);
    }
}
```

LusterWong, 2025-09-08:
> Hi, the default `LV_SYSMON_GET_IDLE` is `lv_os_get_idle_percent` which has a special implementation when `LV_OS_FREERTOS` is enabled.
> 
> Either set `LV_SYSMON_GET_IDLE` to `lv_timer_get_idle` to use the LVGL timer as a reference for CPU usage. Or,
> 
> Define `traceTASK_SWITCHED_IN` and `traceTASK_SWITCHED_OUT` as `lv_port_freertos_task_switch_in` and `lv_port_freertos_task_switch_out` respectively. https://github.com/lvgl/lv_port_renesas-ek-rz_a3m/blob/952f355c97eec95e5bd35ede808935409e3a955a/src/User_FreeRTOSConfig.h#L6-L10

Hello, I have tried including lv_conf.h, and enabling LV_USE_OS=LV_OS_FREERTOS as well as defining LV_SYSMON_GET_IDLE=lv_os_get_idle_percent. However, the CPU usage remained at 100%. 

[https://github.com/lvgl/lv_port_renesas-ek-rz_a3m/blob/952f355c97eec95e5bd35ede808935409e3a955a/src/User_FreeRTOSConfig.h#L6-L10](https://github.com/lvgl/lv_port_renesas-ek-rz_a3m/blob/952f355c97eec95e5bd35ede808935409e3a955a/src/User_FreeRTOSConfig.h#L6-L10)
The method described in the website also fails to solve this problem. I have tried this: 

```
void lv_port_freertos_task_switch_in(const char * name);
void lv_port_freertos_task_switch_out(void);

#define traceTASK_SWITCHED_IN()   lv_port_freertos_task_switch_in(pxCurrentTCB->pcTaskName);
#define traceTASK_SWITCHED_OUT()  lv_port_freertos_task_switch_out();

void my_disp_flush( lv_display_t *disp, const lv_area_t *area, uint8_t * px_map)
{
    ...
    lv_display_flush_ready(disp);
}

void setup(void)
{
    ...
    lcd.init();
    ....
    lv_display_set_buffers(disp, buf0, buf1, drawBufSize, LV_DISPLAY_RENDER_MODE_PARTIAL);
    ....
    lv_demo_benchmark();
}

void loop(void)
{
    lv_timer_handler(); /* let the GUI do its work */
    vTaskDelay(pdMS_TO_TICKS(5)); /* let this time pass */
}
```

Compilation warning message: traceTASK_SWITCHED_IN and traceTASK_SWITCHED_OUT have been redefined, and the CPU usage is still 100%. I suspect that the FreeRTOS for ESP32 has been successfully compiled under Arduino. These two macros have already been confirmed, so there is no way to redefine them.

liamHowatt, 2025-09-19:
It's really a difficult case because FreeRTOS is baked into ESP IDF and I'm not sure how it can be customized. Do you have access to a file called User_FreeRTOSConfig.h? That's where it needs to be done, if so. #including lv_conf.h is not the way to do it.

An immediate solution is to simply set `LV_SYSMON_GET_IDLE` to `lv_timer_get_idle`. This is done in lv_conf.h directly.

lingyunya, 2025-11-08:
> > 
> > I tried the above methods with esp-idf, but none of them worked. The only solution is to set LV_USE_OS to LV_OS_NONE.
> 
> Try adding this to `CMakeLists.txt`
> 
> ```cmake
> add_compile_definitions(CONFIG_LV_SYSMON_GET_IDLE=lv_timer_get_idle)
> ```

This operation can indeed solve the problem. Thank you.

closed 2026-01-22 by lvgl-bot