lvgl/lvgl #8849: When Using Compressed Font Files my Board Crashes

lapowels, 2025-09-09:
### LVGL version

v9.3.0

### Platform

Arduino IDE
Hardware: Arduino Giga R1 WiFi with Giga Display Shield

### What happened?

I boiled my program down to a simple sketch that prints the letter “A” and two seconds later it prints “AA” on the GIGA Display Shield. When I build my sketch using the uncompressed file and LV_USE_FONT_COMPRESSED as 0, the program runs as expected

But when i change the LV_USE_FONT_COMPRESSED to 1 (contained within the "lv_config" file)  and use the Compressed font file, I get the “Red Blinking Death” light flashing on my Arduino Giga R1 Wifi. 

Below I have included My Arduino Sketch and attached the Compressed Font file.  Thanks for your help.

Also, another person on the Arduino Forum tried this on different hardware and his worked.  His environment was:
Board: ESP32 2432S028R (AKA "CYD")
Platform: esp32 by Espressif Systems version 3.3.0
LVGL: version 9.3.0
#define LV_MEM_SIZE (64 * 1024U)
My code based on: [LVGL_Arduino](https://github.com/lvgl/lvgl/tree/master/examples/arduino/LVGL_Arduino)
with [LovyanGFX](https://github.com/lovyan03/LovyanGFX) version 1.2.7 instead of TFT_eSPI

Since he used my same sketch but different Arduino Libraries to support his board, I am thinking that the Arduino Libraries that support my board have issues.

### How to reproduce?

#include "Arduino_H7_Video.h"
#include "Arduino_GigaDisplayTouch.h"

#include "lvgl.h"

Arduino_H7_Video          Display(800, 480, GigaDisplayShield); /* Arduino_H7_Video Display(1024, 768, USBCVideo); */
Arduino_GigaDisplayTouch  TouchDetector;

#define DelayTime 2000
enum LoopModes {LoadingDatabase, ProcessingIncomingCall, Idle}; 
static lv_obj_t * ProgStatusMsg;
static lv_obj_t * SplashPage;
static lv_obj_t* ProgStatus;

void setup() {
  Serial.begin(115200);

  Display.begin();
  TouchDetector.begin();
  lv_init();

  lv_obj_t * SplashPage = lv_obj_create(lv_scr_act());
  lv_obj_set_size(SplashPage, Display.width(), Display.height());

  lv_obj_t * label;
  lv_obj_t * obj;

  LV_FONT_DECLARE(Hawaiian_Size40_4BPP);
  ProgStatus = lv_label_create(SplashPage);

  lv_obj_set_pos(ProgStatus,0,0);
  lv_obj_set_style_text_font(ProgStatus,&Hawaiian_Size40_4BPP, 0);

}

void loop() { 
  static enum LoopModes ProcessingMode = LoadingDatabase;
  static unsigned long time_now = 0;

  if(ProcessingMode == LoadingDatabase) {
      //lv_label_set_text_fmt
      lv_label_set_text(ProgStatus, "A");
      ProcessingMode = ProcessingIncomingCall;
  } else if(ProcessingMode == ProcessingIncomingCall) {
      lv_label_set_text(ProgStatus, "AA");
      ProcessingMode = Idle;
  } 
  time_now = millis();
  while(millis() < time_now + DelayTime){
    lv_timer_handler();  /* Feed LVGL engine */
  }
}

[Hawaiian_Size40_4BPPOneCharCompressed.c](https://github.com/user-attachments/files/22236682/Hawaiian_Size40_4BPPOneCharCompressed.c)

uLipe, 2025-10-02:
Sorry just saw this thing today, I will take a look on it.

uLipe, 2025-11-11:
I was able to reproduce this on my end, I'm doing some gdbuging to track down the issue.

cc: @lapowels

closed 2026-01-29 by lvgl-bot