lvgl/lvgl #8897: [Regression] SDL draw function frees heap created by SW draw functions

ReimuNotMoe, 2025-09-16:
### LVGL version

7843dd3f01c3105c62c50e9cdfef226fd0c6ebc0

### Platform

Linux SDL

### What happened?

[image]

A quick debug shows that when `draw_from_cached_texture` is called, the `u->task_act` contains a `draw_dsc` which belongs to the software renderer: the `draw_dsc` was initialized by the line `(uint8_t *)new_task + LV_ALIGN_UP(sizeof(lv_draw_task_t), 8);` in `lv_draw_add_task()` in `lv_draw.c`. `sizeof(lv_draw_task_t)` being equal to `120` also confirms this.

This bug doesn't exist in `4f2fd4b54950d5efcee9070fbb92d0005236d2ee`.

### How to reproduce?

```c
#define LV_USE_DRAW_SDL 1
#define LV_USE_SDL 1
#define LV_SDL_ACCELERATED 1
```

And create a `lv_menu` with sidebar, alongside with some entries.

kisvegabor, 2025-09-17:
Thank you for reporting it. Please add copy-paste-able code snippet to reproduce it.

matteo-zeggiotti-one, 2025-11-07:
LVGL version: 9.4.0

Platform: Windows

[image]

```
lv_tlsf_free(void * tlsf, const void * ptr) Line 1166 (lvgl\src\stdlib\builtin\lv_tlsf.c:1166)
lv_free_core(void * p) Line 205 (lvgl\src\stdlib\builtin\lv_mem_core_builtin.c:205)
lv_free(void * data) Line 136 (lvgl\src\stdlib\lv_mem.c:136)
sdl_texture_cache_free_cb(cache_data_t * cached_data, void * user_data) Line 79 (lvgl\src\draw\sdl\lv_draw_sdl.c:79)
lv_cache_acquire_or_create(_lv_cache_t * cache, const void * key, void * user_data) Line 187 (lvgl\src\misc\cache\lv_cache.c:187)
draw_from_cached_texture(lv_draw_sdl_unit_t * u) Line 429 (lvgl\src\draw\sdl\lv_draw_sdl.c:429)
execute_drawing(lv_draw_sdl_unit_t * u) Line 509 (lvgl\src\draw\sdl\lv_draw_sdl.c:509)
dispatch(_lv_draw_unit_t * draw_unit, _lv_layer_t * layer) Line 166 (lvgl\src\draw\sdl\lv_draw_sdl.c:166)
lv_draw_dispatch_layer(_lv_display_t * disp, _lv_layer_t * layer) Line 284 (lvgl\src\draw\lv_draw.c:284)
lv_draw_dispatch() Line 222 (lvgl\src\draw\lv_draw.c:222)
lv_draw_finalize_task_creation(_lv_layer_t * layer, _lv_draw_task_t * t) Line 173 (lvgl\src\draw\lv_draw.c:173)
lv_draw_mask_rect(_lv_layer_t * layer, const _lv_draw_mask_rect_dsc_t * dsc) Line 73 (lvgl\src\draw\lv_draw_mask.c:73)
draw_indic(_lv_event_t * e) Line 600 (lvgl\src\widgets\bar\lv_bar.c:600)
lv_bar_event(const _lv_obj_class_t * class_p, _lv_event_t * e) Line 663 (lvgl\src\widgets\bar\lv_bar.c:663)
lv_obj_event_base(const _lv_obj_class_t * class_p, _lv_event_t * e) Line 95 (lvgl\src\core\lv_obj_event.c:95)
lv_slider_event(const _lv_obj_class_t * class_p, _lv_event_t * e) Line 271 (lvgl\src\widgets\slider\lv_slider.c:271)
lv_obj_event_base(const _lv_obj_class_t * class_p, _lv_event_t * e) Line 95 (lvgl\src\core\lv_obj_event.c:95)
event_send_core(_lv_event_t * e) Line 386 (lvgl\src\core\lv_obj_event.c:386)
lv_obj_send_event(_lv_obj_t * obj, lv_event_code_t event_code, void * param) Line 69 (lvgl\src\core\lv_obj_event.c:69)
lv_obj_redraw(_lv_layer_t * layer, _lv_obj_t * obj) Line 126 (lvgl\src\core\lv_refr.c:126)
lv_obj_refr(_lv_layer_t * layer, _lv_obj_t * obj) Line 511 (lvgl\src\core\lv_refr.c:511)
lv_obj_redraw(_lv_layer_t * layer, _lv_obj_t * obj) Line 175 (lvgl\src\core\lv_refr.c:175)
lv_obj_refr(_lv_layer_t * layer, _lv_obj_t * obj) Line 511 (lvgl\src\core\lv_refr.c:511)
lv_obj_redraw(_lv_layer_t * layer, _lv_obj_t * obj) Line 175 (lvgl\src\core\lv_refr.c:175)
lv_obj_refr(_lv_layer_t * layer, _lv_obj_t * obj) Line 511 (lvgl\src\core\lv_refr.c:511)
refr_obj_and_children(_lv_layer_t * layer, _lv_obj_t * top_obj) Line 1097 (lvgl\src\core\lv_refr.c:1097)
refr_configured_layer(_lv_layer_t * layer) Line 1061 (lvgl\src\core\lv_refr.c:1061)
refr_area(const lv_area_t * area_p, int y_offset) Line 896 (lvgl\src\core\lv_refr.c:896)
refr_invalid_areas() Line 819 (lvgl\src\core\lv_refr.c:819)
lv_display_refr_timer(_lv_timer_t * tmr) Line 415 (lvgl\src\core\lv_refr.c:415)
lv_timer_exec(_lv_timer_t * timer) Line 327 (lvgl\src\misc\lv_timer.c:327)
lv_timer_handler() Line 107 (lvgl\src\misc\lv_timer.c:107)
```

SakiiCode, 2025-11-15:
If you enable `LV_USE_ASSERT_MEM_INTEGRITY`, that also catches it

SakiiCode, 2025-11-24:
So you create an `*lv_display_t` with `lv_display_create()` and then an `*lv_draw_buf_t` with `lv_draw_buf_create()`.

Then register the buffer to the display with `lv_display_set_draw_buffers(*lv_display_t, *lv_draw_buf_t, NULL)`

Previously I was using `lv_display_set_buffers()` which accept `void*` pointers instead of `*lv_draw_buf_t` and they were giving weird memory errors

I am using Rust so can't give you exact C code

HeisensOppings, 2025-12-16:
I also encountered this issue on the release/v9.4.

#### LVGL Version
[release/v9.4](https://github.com/lvgl/lvgl/commits/release/v9.4)

#### Platform
Windows MinGW (Msys2) SDL2-2.30.2

#### Code Snippet
```c
lv_demo_widgets();
// Object causing the issue
// lv_obj_t * gender = lv_dropdown_create(panel2);
// lv_dropdown_set_options_static(gender, "Male\nFemale\nOther");
```

##### Issue Description
The crash happens when clicking the dropdown object to expand it, with a similar stack trace to the original report.

[image]

closed 2026-03-05 by lvgl-bot