lvgl/lvgl #8921: Attempting to drag an object which has been positioned using a centre based align function causes the objects coordinates to become corrupted.

pete-pjb, 2025-09-19:
### LVGL version

Latest Master

### Platform

I use the AMD Vitis development environment which is based on VS code.

### What happened?

If an object is placed on the screen and has a drag callback associated with it... If it has been placed with a centre align based function( eg lv_align_centre() or lv_obj_align(LV_ALIGN_CENTER), it's coordinates are corrupted on drag and it disappears off the screen.  If it is placed with lv_obj_set_pos() all is well.  I would have looked into this myself but don't have time right now, sorry!  I have included a very simple bit of code below to reproduce the problem.

### How to reproduce?

```c

static void drag_handler(lv_event_t* e) {

    lv_obj_t            *obj = (lv_obj_t*)lv_event_get_target(e);
    lv_event_code_t		code = lv_event_get_code(e);

    if (code == LV_EVENT_PRESSING) {

        lv_indev_t* indev = lv_indev_get_act();
        if (indev == NULL)  return;

        lv_point_t vect;
        lv_indev_get_vect(indev, &vect);

        lv_coord_t x = lv_obj_get_x(obj) + vect.x;
        lv_coord_t y = lv_obj_get_y(obj) + vect.y;
        lv_obj_set_pos(obj, x, y);
        lv_obj_update_layout(obj);
    }
}

    lv_obj_t* container = lv_obj_create(lv_scr_act());
    lv_obj_set_size(container, 80, 60);
//   lv_obj_set_pos(container, 10, 10);                     // This works correctly
//   lv_obj_align(container, LV_ALIGN_TOP_LEFT, 10, 10);    // This works correctly
    lv_obj_align(container, LV_ALIGN_CENTER, 10, 10);       // This fails
//   lv_obj_center(container);                              // This fails
    lv_obj_add_event_cb(container, drag_handler, LV_EVENT_PRESSING, NULL);
```

pete-pjb, 2025-09-25:
Update to this issue - it seems an object with any alignment **_except_** LV_ALIGN_TOP_LEFT has this issue.  Stepping through the code after the drag handler is called the coordinates of the object keep doubling which maybe suggests one of the styles is being reapplied over and over?  I don't know enough about the internals of LVGL to say for sure why this is happening...

If you have any thoughts @kisvegabor would be helpful to get a hint from you 🙂

kisvegabor, 2025-09-29:
Hi Pete,

Please take a look at [lv_example_obj_2](https://github.com/lvgl/lvgl/blob/master/examples/widgets/obj/lv_example_obj_2.c).

The trick is to use `lv_obj_get_x_aligned` and `lv_obj_get_y_aligned` to get the coordinates as `lv_obj_get_x/y` returns the coordinates form the top left corner.

closed 2025-09-29 by pete-pjb