lvgl/lvgl #8926: Mixing of GL_RGBA and GL_BGRA in texture formats fails on some drivers

jtrknut, 2025-09-22:
### LVGL version

v9.3.0-431-g671edcd7f

### Platform

I'm using an embedded Linux platform, based on Yocto on the NXP i.MX95 with their binary driver.

I'm using the DRM driver with EGL.

Here are the list of the supported extensons on my platform
[imx95_exts.txt](https://github.com/user-attachments/files/22458945/imx95_exts.txt)

### What happened?

I've encountered an issue when trying to run with OpenGL acceleration on the i.MX95 mali driver.

The glTexImage2D calls fail bacause it does not accept the texture combination of GL_RGBA mixed with GL_BGRA in the format specifiers in glTexImage2D calls.

When looking at the documentation I cannot see the that mix as a supported configuration:
https://registry.khronos.org/OpenGL-Refpages/es3.0/html/glTexImage2D.xhtml
And
https://registry.khronos.org/OpenGL/extensions/EXT/EXT_texture_format_BGRA8888.txt

Changing both the internal and external formats to GL_BGRA fixes the issue on my end.

### How to reproduce?

This can be reproduced by simply enabling the the DRAW_OPENGLES option. There are some other issues with this platform, but they are not directly related to this issue and will be reported in a different issue / pull request.

jtrknut, 2025-10-03:
Thank you for your updated backend.

The new code still has this issue. If I swap to GL_BGRA in both internal and external formats in lv_draw_opengles.c and lv_opengles_texture.c, things works again.

Here's the errors I'm getting:

```
[ERROR] [Error] (1050.614, +1050614)     GLLogCall: [OpenGL Error] (1282) glTexImage2D(GL_TEXTURE_2D, 0, GL_RGBA, w, h, 0, GL_BGRA, GL_UNSIGNED_BYTE, data) lvgl-src/src/draw/opengles/lv_draw_opengles.c:625 lv_opengles_debug.c:51
[ERROR] [Error] (1050.617, +3)   GLLogCall: [OpenGL Error] (1282) glTexImage2D(GL_TEXTURE_2D, 0, GL_RGBA, w, h, 0, GL_BGRA, GL_UNSIGNED_BYTE, data) lvgl-src/src/draw/opengles/lv_draw_opengles.c:625 lv_opengles_debug.c:51
[ERROR] [Error] (1050.618, +1)   GLLogCall: [OpenGL Error] (1282) glTexImage2D(GL_TEXTURE_2D, 0, GL_RGBA, w, h, 0, GL_BGRA, GL_UNSIGNED_BYTE, data) lvgl-src/src/draw/opengles/lv_draw_opengles.c:625 lv_opengles_debug.c:51
```

jtrknut, 2025-10-07:
I would like to add that to inspect the colors, I've used the online image conversion tool to generate an image that I then put on the screen with the lv_image control.

jtrknut, 2025-10-20:
Thanks for looking in to this.

At the moment I just patch the color formats to use GL_BGRA on both internal and external formats and this works for me perfectly.

I'll keep an eye out for the OpenGLES refactor and test that once it arrives.

AndreCostaaa, 2026-01-09:
Hi @jtrknut #9304 got merged and we no longer use BGRA. Closing this issue now, let me know if you encounter any more issues and thanks for the help raising this issue :)