lvgl/lvgl #8963: Cache coherency issue in lv_st_ltdc driver: LTDC (as bus master) reads stale data when D-Cache is enabled

git-siyun, 2025-09-28:
### LVGL version

9.3.0

### Platform

STM32F7/H7

### What happened?

Hello LVGL team,

I’d like to report a rendering issue with the built-in lv_st_ltdc driver on Cortex-M7 MCUs (e.g., STM32H7) when the data cache (D-Cache) is enabled.

The problem occurs in LV_DISPLAY_RENDER_MODE_DIRECT mode: after LVGL finishes drawing into the frame buffer, the display sometimes shows corrupted or outdated content.

Root cause:

The LTDC peripheral acts as a bus master—it reads pixel data directly from memory, without going through the CPU cache. If the frame buffer resides in a cacheable memory region (which is common), the CPU’s writes may only be in the D-Cache and not yet written back to RAM. When LTDC fetches the data, it sees the old (uncached) content, leading to visual glitches.

This is not related to DMA or DMA2D—it happens even in pure CPU rendering, simply because LTDC and the CPU are two independent masters accessing the same memory.

Fix suggestion:

Before telling LTDC to use a newly rendered frame buffer (e.g., before calling HAL_LTDC_SetAddress_NoReload()), the driver should flush the relevant cache lines:

SCB_CleanDCache_by_Addr((uint32_t*)draw_buf_addr, draw_buf_size);
Without this, cache coherency between CPU and LTDC cannot be guaranteed on cache-enabled systems.

Thanks for your great work—and I hope this helps improve robustness on STM32H7/M7 platforms!

kisvegabor, 2025-09-28:
Cc @liamHowatt

git-siyun, 2025-10-01:
Regarding the STM32 project to reproduce the issue, my H7 hardware is currently unavailable, so I cannot provide a validated H7 project. However, I can offer you a validated F7-based project for reference.
**Hardware Notes:**
- F7 vs H7 Similarity: Both F7 and H7 use Cortex-M7 cores with identical cache architecture and memory models
- Issue Reproducibility: This cache coherence issue occurs on all M7 core chips, with consistent behavior between F7 and H7
- Validation Reliability: Results validated on F7 are fully applicable to H7 platforms
**Issue Demonstration Video:**
I've prepared a recorded video demonstrating the problem. Please pay special attention to:
- Display anomalies in the Monthly Target chart
- Tearing or misalignment in the text below the chart
- These display issues are exactly caused by unsynchronized cache data
**Fix Verification Results:**
After applying DeepSeek's recommended cache cleaning solution:
··· c
if(disp->render_mode == LV_DISPLAY_RENDER_MODE_DIRECT) {
    if(lv_display_is_double_buffered(disp) && lv_display_flush_is_last(disp)) {
#if defined(__DCACHE_PRESENT) && (__DCACHE_PRESENT == 1)
        uint32_t color_format = lv_display_get_color_format(disp);
        uint32_t px_size = lv_color_format_get_size(color_format);
        uint32_t buffer_size = disp->hor_res * disp->ver_res * px_size;
        SCB_CleanDCache_by_Addr(px_map, buffer_size);
#endif
        HAL_LTDC_SetAddress_NoReload(&hltdc, (uint32_t)px_map, layer_idx);
        // ... rest of the code
    }
}
···
The display issues completely resolved, with stable and normal rendering of both charts and text.

I'll send you both the F7 project and the demonstration video. You can directly apply the same fix to your H7 platform.

[demo.zip](https://github.com/user-attachments/files/22631414/demo.zip)

https://github.com/user-attachments/assets/c5051d47-f7b2-444b-9b46-b7acf2efe608